OpenC3/cosmos #22: Scriptable TlmGrapher for automatic plot generation

zane-sauer, 2022-07-25:
I found a feature request [here](https://github.com/BallAerospace/COSMOS/issues/1701) on the COSMOS repo to make TlmGrapher load config scriptable. Have there been any updates to the feature linked previously?

jmthomas, 2022-07-25:
TlmGrapher already can save and load a configuration. However it would be nice to make the configuration load scriptable and the configuration store the time ranges (fixed and relative, e.g. 1hr ago) so you could automate the generate of plots. Go to the TlmGrapher URL with the right parameters and autogenerate a plot which could be screenshotted and downloaded.

jmthomas, 2022-07-25:
No additional work yet but knowing someone values this feature will bump it up on our list. :-)

jmthomas, 2022-10-05:
We also need the ability to export graphs as images or PDFs

ryanmelt, 2023-02-16:
This functionality is going to be supported through Grafana in a future release.  Closing this ticket.

closed 2023-02-16 by ryanmelt