OpenC3/cosmos #31: TCP interface

AstreeDeshayes, 2022-07-27:
Hello,

I can not succeed to connect a TCPIP socket to send commands to a python server. I have the following error message : 

HYCUBE_INT: Connection Failed: ECONNREFUSED : Connection refused - connect(2) for 172.17.0.1:8777

Here is my plugin.txt configuration for the interface : 

VARIABLE hycube_target_name HYCUBE

TARGET HYCUBE <%= hycube_target_name %>
INTERFACE <%= hycube_target_name %>_INT tcpip_client_interface.rb 172.17.0.1 8777 8777 nil BURST
  MAP_TARGET <%= hycube_target_name %>

I tried to change 172.17.0.1 with host.docker.internal but it didn't change anything.

Do you know why ?

Thanks,
Astrée Deshayes

ryanmelt, 2022-07-27:
Can you provide more details on your python process?   
Is it on the same computer you are running OpenC3?
Is it listening on 0.0.0.0, or 127.0.0.1?   
Are you sure a local firewall isn't preventing connections?

Also I think your INTERFACE line is missing a timeout value before the nil (usually 10.0 in examples).
INTERFACE <%= hycube_target_name %>_INT tcpip_client_interface.rb 172.17.0.1 8777 8777 10.0 nil BURST

AstreeDeshayes, 2022-08-02:
Thanks for the advice, I solved my problem by increasing timeout and changing some parameters in my code.

closed 2022-08-02 by ryanmelt

MuhammadRamzy, 2025-07-01:
I'm currently using 127.0.0.1:8080 as well, and I had added the port mapping in my docker compose.yaml, but I was still encountering the same issue. Could you please share the specific changes you made and the Python code you used to connect? It would be really helpful to compare and see what might be missing on my end.