OpenC3/cosmos #55: Can't access from other than localhost

rmorang, 2022-08-09:
I’m trying to deploy a telemetry server for my Formula Student team, and I came across OpenC3. I became fascinated by it and wanted to give a try. After becoming familiar with it in Docker Desktop, I wanted to build it on a Raspberry Pi 4B with 8GB of RAM. I followed your guide, but when it came to connecting to http://openc3.local:2900/ I wasn’t able to. I’ve checked all the configuration files to see if the port was not exposed correctly but wasn’t able to find any evidence. Is there something I’m missing? Do I need to modify any file?

ryanmelt, 2022-08-09:
Let me update my Pi and see if I can recreate

ryanmelt, 2022-08-09:
What Operating system is your browser running on?   If Windows, you might have to take some steps to get .local domains to work

rmorang, 2022-08-09:
I'm running Mac OS Monterey with Safari

rmorang, 2022-08-09:
Do I need to modify anything in docker compose? There are some commented lines:
``openc3-traefik:
    image: "${OPENC3_REGISTRY}/openc3inc/openc3-traefik:${OPENC3_TAG}"
    volumes:
      - "./cacert.pem:/devel/cacert.pem:z"
      - "./openc3-traefik/traefik.yaml:/etc/traefik/traefik.yaml:z"
      # - "./openc3-traefik/traefik-allow-http.yaml:/etc/traefik/traefik.yaml"
      # - "./openc3-traefik/traefik-ssl.yaml:/etc/traefik/traefik.yaml"
      # - "./openc3-traefik/traefik-letsencrypt.yaml:/etc/traefik/traefik.yaml"
      # - "./openc3-traefik/cert.key:/etc/traefik/cert.key"
      # - `"./openc3-traefik/cert.crt:/etc/traefik/cert.crt"```

rmorang, 2022-08-09:
Managed to fix it:

`ports:
      -  2900:80
      -  2943:443```

closed 2022-08-09 by rmorang

andrewvzorn, 2022-08-18:
hi rmorang. I am having the same problem that you had. I can only access the web interface directly from the raspberry pi. I cannot access it using http://openc3.local:2900/ from my laptop on the same network. Can you explain your solution?

rmorang, 2022-08-22:
> hi rmorang. I am having the same problem that you had. I can only access the web interface directly from the raspberry pi. I cannot access it using http://openc3.local:2900/ from my laptop on the same network. Can you explain your solution?

Sorry for not answering before. First of all, are you sure your host name is openc3? If that's so check this thread thoroughly as I describe how I managed to fix it.