OpenC3/cosmos #56: Packet 'EXAMPLE STATUS' does not exist

ryanmelt, 2022-08-09:
The CVT contains no data for the EXAMPLE STATUS packet in the demo by default, which is a causing a 500 error, packet does not exist in Packet Viewer

jmthomas, 2022-08-11:
@ryanmelt I simply see a blank page with all the values set to `null`. I do not get the 500 error. Can you reproduce this on master?

ryanmelt, 2022-08-12:
Figured this out.  This is happening because the Current Value Table (CVT) is stored in the Ephemeral Redis container.   As soon as that container restarts for any reason, the packet full of nil values is lost and no longer exists.  @jmthomas

ryanmelt, 2022-08-12:
Can be recreated immediately with: docker restart openc3-openc3-redis-ephemeral-1

PR #68 (2022-08-13): Move CvtModel back to Persistent Redis Store
closes #56
closed 2022-08-13 by ryanmelt