OpenC3/cosmos #57: Serial interface in linux

rmorang, 2022-08-09:
I want to read fron an arduino  serial port in a raspberry pi. I know the --device flag gives access to the port. However I don't in which image should i add this flag and how to do it.

ryanmelt, 2022-08-09:
You need to add it to the operator container

ryanmelt, 2022-08-09:
Something like this with the correct device name:

   devices:
     - "/dev/ttyUSB0:/dev/ttyUSB0"

rmorang, 2022-08-09:
I tried to pass it through and got this message:

BLINK_INT: ENOENT : No such file or directory @ rb_sysopen - /dev/ttyACM0 /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/io/posix_serial_driver.rb:55:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/io/posix_serial_driver.rb:55:in `open' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/io/posix_serial_driver.rb:55:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/io/serial_driver.rb:67:in `new' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/io/serial_driver.rb:67:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/streams/serial_stream.rb:74:in `new' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/streams/serial_stream.rb:74:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/interfaces/serial_interface.rb:68:in `new' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/interfaces/serial_interface.rb:68:in `connect' interface_microservice.rb:459:in `connect' interface_microservice.rb:314:in `block in run' interface_microservice.rb:312:in `synchronize' interface_microservice.rb:312:in `run' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.8.pre.beta0/lib/openc3/microservices/microservice.rb:47:in `run' interface_microservice.rb:532:in `<main>'

ryanmelt, 2022-08-09:
Doesn't sound like it made it into the container.   If you do this:
docker exec -it <container id> sh

You can ls /dev and see if it is there

rmorang, 2022-08-09:
/openc3/lib/openc3/operators $ ls /dev
fd       full     mqueue   null     ptmx     pts      random   shm      stderr   stdin    stdout   tty      ttyACM0  urandom  zero

rmorang, 2022-08-09:
I added the user pi to the group dialout and now this message appears:

BLINK_INT: EACCES : Permission denied @ rb_sysopen - /dev/ttyACM0 /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/io/posix_serial_driver.rb:55:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/io/posix_serial_driver.rb:55:in `open' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/io/posix_serial_driver.rb:55:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/io/serial_driver.rb:67:in `new' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/io/serial_driver.rb:67:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/streams/serial_stream.rb:74:in `new' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/streams/serial_stream.rb:74:in `initialize' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/interfaces/serial_interface.rb:68:in `new' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/interfaces/serial_interface.rb:68:in `connect' interface_microservice.rb:459:in `connect' interface_microservice.rb:314:in `block in run' interface_microservice.rb:312:in `synchronize' interface_microservice.rb:312:in `run' /usr/lib/ruby/gems/3.0.0/gems/openc3-5.0.7/lib/openc3/microservices/microservice.rb:47:in `run' interface_microservice.rb:532:in `<main>'

rmorang, 2022-08-09:
BLINK_INT: Connection Failed: EACCES : Permission denied @ rb_sysopen - /dev/ttyACM0

ryanmelt, 2022-08-09:
Try adding to the operator container: 
privileged: true

rmorang, 2022-08-09:
Added it to the compose.yml but nothing changed

ryanmelt, 2022-08-10:
Maybe:
sudo chmod 666 /dev/ttyACM0

rmorang, 2022-08-10:
Got it! Instead of devices y had to add "/dev/ttyACM0:/dev/ttyACM0" as a volume and then chmod 666 /dev/ttyACM0

closed 2022-08-10 by rmorang