OpenC3/cosmos #60: Command packets displaying unknown behavior

tj-scherer, 2022-08-10:
When I create a CCSDS command packet, it seems to be rearranging the fields out of the specified order and adding a random offset in the middle.

***plugin.txt***
```tcl
# Set VARIABLEs here to allow variation in your plugin
# See https://openc3.com/docs/v5/plugins for more information
VARIABLE rce_tvs_target_name RCE_TVS

# Modify this according to your actual target connection
# See https://openc3.com/docs/v5/interfaces for more information
TARGET RCE_TVS <%= rce_tvs_target_name %>
INTERFACE <%= rce_tvs_target_name %>_INT tcpip_client_interface.rb host.docker.internal 8080 8080 10.0 nil LENGTH 96 16 15 1 BIG_ENDIAN 8 0xdeadbeefdeadbeef 65536 true nil
  MAP_TARGET <%= rce_tvs_target_name %>
```

***_ccsds_tlm.txt***
```tcl
	APPEND_ITEM    CCSDSVER      3  UINT             "CCSDS PACKET VERSION NUMBER (SEE CCSDS 133.0-B-1)"
		STATE V1 0
	APPEND_ITEM    CCSDSTYPE     1  UINT             "CCSDS PACKET TYPE (COMMAND OR TELEMETRY)"
		STATE TLM 0
		STATE CMD 1
	APPEND_ITEM    CCSDSSHF      1  UINT             "CCSDS SECONDARY HEADER FLAG"
		STATE FALSE 0
		STATE TRUE 1
	APPEND_ID_ITEM CCSDSAPID     11 UINT <%= apid %> "CCSDS APPLICATION PROCESS ID"
	APPEND_ITEM    CCSDSSEQFLAGS 2  UINT             "CCSDS SEQUENCE FLAGS"
		STATE CONT 0
		STATE FIRST 1
		STATE LAST 2
		STATE NOGROUP 3
	APPEND_ITEM    CCSDSSEQCNT   14 UINT             "CCSDS PACKET SEQUENCE COUNT"
	APPEND_ITEM    CCSDSLENGTH   16 UINT             "CCSDS PACKET DATA LENGTH"
```

***commands_cmd.txt***
```tcl
COMMAND <%= target_name %> TEST BIG_ENDIAN "Test"
	<%= render "_ccsds_cmd.txt", locals: {apid: 1031} %>
	
	APPEND_PARAMETER TEST 16 UINT MIN MAX 0xdead "Test"
```

[image]

That produces the handbook shown in the image. Obviously, this is both out of order (all of the CCSDS fields should be rendered before the TEST field) and also shows a large bit offset being introduced without reason.

- OS: Windows 10
- OpenC3 Version 5.0.7-beta0
- Browser
  -- [x] Google Chrome

ryanmelt, 2022-08-10:
You included _ccsds_tlm.txt not your _ccsds_cmd.txt file.  Can we see that?

tj-scherer, 2022-08-11:
Sorry about that, original post has been updated.

jmthomas, 2022-08-11:
Your CCSDSTYPE parameter is at bit offset 559! So actually everything is in order except for some reason you have a huge offset. Our demo defines parameters as absolute offsets (not APPEND) so I'll try to reproduce.

tj-scherer, 2022-08-11:
Yes, the issue is that I am using APPEND_PARAMETER which appears to be introducing this offset erroneously. The order of the fields also makes no sense. All of the CCSDS fields are rendered in the same file, then the TEST field is appended afterwards. There is no apparent cause for the TEST field to be interleaved, yet it is. I will try to do some testing without using sync markers, because that is when I first noticed the command packets acting strangely. I just don't see another explanation for the random order and random bit offsets.

tj-scherer, 2022-08-11:
I'm actually coming across some new information. It looks like the packet was getting jumbled up by lines that were commented out. Is this a known issue with the parser? The following was included in ***commands_cmd.txt*** and the issue disappeared when it was all removed:

```tcl
# COMMAND <%= target_name %> DELPHICORE_HARDWARE_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 0} %>

# COMMAND <%= target_name %> DELPHICORE_IOMUX_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 1} %>

# COMMAND <%= target_name %> DELPHICORE_IOINTERFACE_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 2} %>

# COMMAND <%= target_name %> DELPHICORE_PULSEGENCHECK_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 3} %>

# COMMAND <%= target_name %> DELPHICORE_TSDI_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 4} %>

# COMMAND <%= target_name %> DELPHICORE_MFG_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 5} %>

# COMMAND <%= target_name %> DELPHICORE_MFC_CMD BIG_ENDIAN "Description"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 6} %>

# COMMAND <%= target_name %> DELPHICORE_SPACEWIRE_CMD BIG_ENDIAN "DelphiCore GetTelemetry()->SpwPathTlm"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 7} %>

# 	APPEND_PARAMETER QUIET 8 UINT 0 1 1 "Get quietly"

# bool SendPacket(uint8 PortNum, SwPacket *Packet, uint32 *BytesSent = NULL, float Timeout = 0, bool Blocking = TRUE, bool Quiet = true);
# COMMAND <%= target_name %> DELPHICORE_SPACEWIRE_SENDPACKET BIG_ENDIAN "DelphiCore SendPacket"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 8} %>

# 	APPEND_PARAMETER PORTNUM 32 UINT 0 MAX 0 "SpaceWire port select"
# 	APPEND_PARAMETER TIMEOUT 32 FLOAT 0 MAX 0 "Send timeout"
# 	APPEND_PARAMETER BLOCKING 8 UINT 0 1 1 "Blocking send"
# 	APPEND_PARAMETER QUIET 8 UINT 0 1 1 "Send quietly"
# 	APPEND_PARAMETER PACKET 0 BLOCK 0 "SpaceWire packet to send"

# bool SetSpwNodeEnable(uint8 PortNum, bool State);
# COMMAND <%= target_name %> DELPHICORE_SPACEWIRE_SETSPWNODEENABLE BIG_ENDIAN "DelphiCore SetSpwNodeEnable()"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 9} %>

# bool SetSpwTxDataRate(uint8 PortNum, uint32 RequestRate, uint32* ActualRate);
# COMMAND <%= target_name %> DELPHICORE_SPACEWIRE_SETSPWTXDATARATE BIG_ENDIAN "DelphiCore SetSpwTxDataRate()"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 10} %>

# bool SetSpWLocalLoopbackMode(uint8 PortNum, bool Flag, bool Quiet = true);
# COMMAND <%= target_name %> DELPHICORE_SPACEWIRE_SETSPWLOCALLOOPBACKMODE BIG_ENDIAN "DelphiCore SetSpwLocalLoopbackMode()"
# 	<%= render "_ccsds_cmd.txt", locals: {apid: 11} %>
```

ryanmelt, 2022-08-11:
That is just how it works.  Substitution occurs even in comments.  You’ll need to not leave any renders commented out

closed 2022-08-11 by tj-scherer