OpenC3/cosmos #72: Differences between COSMOS and OpenC3

keck-in-space, 2022-08-15:
Hello,

I am curious what option I should choose between OpenC3 and COSMOS v5. Is there a list of differences anywhere?

Thank you.

ryanmelt, 2022-08-15:
OpenC3 is the continuation of COSMOS 5, by its creators.   We've forked COSMOS and are continuing it full-time as a new company, OpenC3.   

The current differences are the changes we've put into the 3 releases we've done so far.   

You can see the release notes here: https://github.com/OpenC3/openc3/releases

Also, we've just finished the first version of OpenC3 Enterprise Edition - a paid product that adds multi-user support, RBAC, and scopes, with a lot more to come.

keck-in-space, 2022-08-16:
Thanks Ryan. Will COSMOS v5 be developed further, or only OpenC3?

jmthomas, 2022-08-16:
The 2 authors of COSMOS are now at OpenC3 and we can only speak to what we're doing with OpenC3. Please contact us at support@openc3.com and perhaps we can setup a call to discuss further.

closed 2022-08-16 by jmthomas

keck-in-space, 2022-08-16:
Got it. Thanks so much.