OpenC3/cosmos #76: Support Grafana

jmthomas, 2022-08-15:
Perhaps push data to InfluxDB with Grafana on top or directly stream data to Grafana

tsjordan-eng, 2022-08-26:
It looks like it support Prometheus and Grafana: https://openc3.com/docs/v5/monitoring#grafana.
I haven't gotten it working yet, though.

jmthomas, 2022-10-05:
Can Graphana support plotting states / enumerations. Can it plot equations based on telemetry? Does it support X vs Y plots?Histogram plots? Can it show units? Can it export as image or PDF?

tsjordan-eng, 2022-11-22:
Is Grafana an enterprise only feature? That link gives a 404

ryanmelt, 2022-11-22:
At least for now, pushing data out to InfluxDb and subsequent Grafana support is going to be Enterprise only, so that we can properly support it.

We are also looking into doing a direct Grafana plugin for COSMOS.

PratikAher20, 2023-01-20:
Eagerly waiting for Open source Grafana Support from COSMOS Team!!

jmthomas, 2023-06-02:
Grafana support is being performed in this repo: https://github.com/OpenC3/grafana-cosmos-datasource. Status will be tracked there. Currently the plugin has been submitted to Grafana for review.

closed 2023-06-02 by jmthomas