OpenC3/cosmos #78: Multiple variable sized items

jmthomas, 2022-08-16:
A fairly common pattern is to have multiple length followed by variable length field within a packet. Come up with a probably inefficient, but working method of specifying this pattern.

apogeecmb, 2023-03-16:
I'm just curious if there has been any progress on implementing a feature like this.  I'm currently trying to implement a packet that follows this pattern and coming up blank on how I might do it.  We have a set of data values that can be repeated X number of times, and I'm trying to figure out how I might parse that out and have it displayed correctly in OpenC3.

I noticed in the example at the bottom of the Telemetry documentation page (https://openc3.com/docs/v5/telemetry) the MACRO_APPEND_START and MACRO_APPEND_END keywords.  I was wondering if something like this might work.

```
ITEM NUMBER_OF_REPEATED_ITEMS 0 8 UINT "Number of repeated items"
ITEM DATA_BLOCK 8 0 BLOCK "Block of data"
MACRO_APPEND_START 1 NUMBER_OF_REPEATED_ITEMS 
    ITEM DERIVED_ENTRY_1 0 0 DERIVED "First item in repeated block"
        READ_CONVERSION ...
    ITEM DERIVED_ENTRY_2 0 0 DERIVED "Second item in repeated block"
        READ_CONVERSION ...
MACRO_APPEND_END
```

I didn't see any documentation on how the MACRO_APPEND_START keyword works, so I'm just guessing at this point.  Thanks for any guidance you can provide.

ryanmelt, 2023-03-17:
I haven't worked on this yet, but it is going to be implemented using the new Accessor functionality.   We can create an accessor that can handle multiple variable sized items as long as there is an associated length value for each item.

tj-scherer, 2023-03-21:
This would be incredibly useful!

tsjordan-eng, 2023-05-26:
COSMOS could also support variable-length strings by reading bytes until NULL, like a C-string.

I think most telemetry would encode a variable-length array as a size followed by an array. 
I've started looking into the binary accessor. It seems that the `item.array_size` could be overloaded so that 0 would retain its current meaning (all bytes until end of packet), but negative sizes could be the `array_size` length.
So the definition could be:
```
  APPEND_ARRAY_PARAMETER ITEM5 8 UINT -16 "Array Item"
```
for an array with a `array_size` field 16 bits long followed by `array_size` 8-bit unsigned integers.

It doesn't have the same flexibility as the LENGTH protocol, but it could be overcome by creating a custom protocol to format the incoming packet this way.

Where does this issue land in terms of priority?

ryanmelt, 2023-05-29:
Negative sizes already have a meaning (fill the rest of the packet except however many bits).

This issue currently falls behind Python support which is going to take most of the summer.  Probably this fall.

PR #1436 (2024-07-29): Multi variable bit size
closes #78