OpenC3/cosmos #79: Increase string byte size

AstreeDeshayes, 2022-08-16:
Hello,

I would like to send this : "2022-08-16T08:32:22.538Z", as a string, in an command. However I got the following error message when trying to send it : 
Status: Error sending HYCUBE_TIME FORCE_EMPTY_MEMORY due to RuntimeError: value of 24 bytes does not fit into 7 bytes for data_type STRING

How can I increase the byte size of the string ?

This is how my command is defined in the cmd.txt file: 

  APPEND_PARAMETER    Date      56     STRING  "2022-08-16T08:32:22.538Z" "Date in UTC format"

Thanks,
Astrée Deshayes

jmthomas, 2022-08-16:
The parameter definition is in bits, thus your definition of 56 bits is 7 bytes which is 7 characters. Multiple the number of characters you want by 8 and change 56 to that number.

AstreeDeshayes, 2022-08-17:
Right thanks!

closed 2022-08-18 by jmthomas