OpenC3/cosmos #80: python api A0 being encoded as C2A0

thesamprice, 2022-08-16:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
in the python api
```
send_raw('HW_INT',b"\xa0")
```
On a C udp interface i am seeing 2 bytes.
```
"\xc2\xa0"
```
Somewhere between the python api and my C interface, string encoding is being applied to byte data.
Im unsure how to stop this, could a test for this be added to the system?
A may be using an older version of openc3....

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS: [e.g. Windows 10, Mac OS 10.13]
- OpenC3 Version [e.g. 5.0.6]
- Browser
  -- [] Internet Explorer (IE)
  -- [] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

thesamprice, 2022-08-18:
I wrote a workaround in my code to deal with this temporarily. 

If I use the send_raw interface in python i send a 0x1 byte first.
and the C code sees this and then fixes the unicode data.
from what i remember 0 to 0x80 is encoded properly.
then it prefixes 0xc2 for the next 0x80 bytes, followed by 0xc3 for the following ones.

Below is my workaround code for this if that helps.
```c
void Fix_Unicode_Data(char * _data, int _bytes, int * _new_bytes){
    /* First byte marks that data is unicoded */

    int src_i;
    int dest_i = 0;
    char byte;
    int new_bytes = 0;
    for(src_i = 1;src_i<_bytes; src_i++){
        if(_data[src_i] == (char) 0xc2){
            src_i++;
            byte = _data[src_i];
        }else if(_data[src_i] == (char) 0xc3){
            src_i++;
            byte = _data[src_i];
            byte += 64;
        }else{
            byte = _data[src_i];
        }
        _data[dest_i] = byte;
        dest_i++;
        new_bytes++;
    }
    *_new_bytes = new_bytes;
}
```

jmthomas, 2022-08-18:
Thanks @thesamprice! Can you reproduce on OpenC3 or is this specific to old COSMOS code?

thesamprice, 2022-08-18:
@jmthomas I dont have much free time to test on OpenC3.  were using cosmos 4.5.2.

I would recommend changing test framework just change the test data from `0x00` to `0xa0`

ryanmelt, 2023-02-13:
We'll be redoing all the python apis starting next month. Closing this for now.

closed 2023-02-13 by ryanmelt

thesamprice, 2023-02-14:
@ryanmelt If possible we should tagup about your python api.

ryanmelt, 2023-02-14:
Email support@openc3.com and we can set something up.