OpenC3/cosmos #84: Pass script parameters from telemetry screens

CharlotteMorrison, 2022-08-17:
Hello,

I would like to be able to pass parameters from a telemetry screen to a script like in Cosmos 4.  I am currently in the process of updating a Cosmos4 project and this functionality is used frequently in the old repo.  Please let me know if this feature is something you are planning to implement, or if you have a suggested method to do this currently.

Thanks again, 
Charlotte

ryanmelt, 2022-08-18:
Were you passing parameters before using ARGV?   
We support passing parameters in OpenC3 using environment variables, though I need to check and see if that works from telemetry screens.  I'll try it out and add some instructions here in the next few days.

CharlotteMorrison, 2022-08-18:
We were passing variables through a couple of methods, including ARGV.  I am not picky how they are passed, just need to get a few things from the screens to scripts to 'clone' the previous screen functionality.

jwestra-carrastro, 2022-11-02:
> Were you passing parameters before using ARGV? We support passing parameters in OpenC3 using environment variables, though I need to check and see if that works from telemetry screens. I'll try it out and add some instructions here in the next few days.

Where was this posted? I would like to know how to pass environmental variables to scripts from a screen.

PR #284 (2022-11-25): Tlm viewer env to scripts
closes #159
closes #84
closed 2022-11-25 by jmthomas