OpenC3/cosmos #88: How do I send a command to an instrument from arbitrary Ruby code?

jhayes-apl, 2022-08-22:
I know how to send a command to an instrument from a script. How
do I send a command from arbitrary Ruby code? For example, I want
to send a command to the instrument when COSMOS receives a specific
telemetry packet. I am trying to use a PROCESSOR:
<PRE>
TELEMETRY LVX TIME_REQUEST BIG_ENDIAN "Time request"
  <%= render "_ccsds_tlm.txt", locals: {id: 0x00} %>
  <%= render "_timetag_tlm.txt" %>
  # Keyword      Name           BitSize Type   ID      Description
  APPEND_ITEM    CRC            16      UINT           "CRC"
    FORMAT_STRING "0x%04x"
  PROCESSOR TIMEREQRESP time_service.rb
</PRE>

In time_service.rb, I tried to use syntax similar to what I use in
scripts. I just get an error about undefined method. (After cmd
didn't work, I was just guessing.)
<PRE>
module Cosmos
  class TimeService < Processor
    ... details suppressed ...
    def call(packet, buffer)
# TBD: doesn't work: tried cmd, api.cmd, Api.cmd, Cosmos.cmd, Cosmos.Api.cmd
      Cosmos.Api.cmd("LVX TIME_RESPONSE with RCV_TIME 1, XMT_TIME 2")
</PRE>

I have also posted this question to COSMOS discussions.  I am using COSMOS 5.0.3.

jmthomas, 2022-08-22:
In time_service.rb you need to require the api and include in:

    require 'openc3/api/api'
    module OpenC3
      class TimeService < Processor
        include Api
      end
    end

jhayes-apl, 2022-08-23:
That works.  Thanks.

closed 2022-08-24 by ryanmelt