OpenC3/cosmos #89: Command sends from Command Sender, but re-executing fails

tj-scherer, 2022-08-22:
I am trying to issue a command containing a BLOCK data type. My ultimate goal is to pass a file to this field of the packet. I was running into issues doing so in the script, so I shifted gears to the Command Sender. When I fill the Command Sender with the contents of the file, the command sends and I can see it being passed through the socket.

[image]

The issue is, when I use the command history below (as well as performing the action in a script), I get the following error:

```
Error sending RCE_TVS RMAP_WRITE_CMD due to RuntimeError: Missing comma in command parameters: RCE_TVS RMAP_WRITE_CMD with TARGET 0, ADDRESS 0, EXTADDR 0, VERIFY 0, ACK 0, AUTOINCR 1, TIMEOUT 0, VERBOSE 0, DATA REQUIRE rce_tvs.rb REQUIRE arr_to_dec_conversion.rb REQUIRE dec_to_bin_conversion.rb # Ignored Parameters # IGNORE_PARAMETER parameter_name IGNORE_PARAMETER CCSDSVER IGNORE_PARAMETER CCSDSTYPE IGNORE_PARAMETER CCSDSSHF IGNORE_PARAMETER CCSDSAPID IGNORE_PARAMETER CCSDSSEQFLAGS IGNORE_PARAMETER CCSDSSEQCNT IGNORE_PARAMETER CCSDSLENGTH # Ignored Items # IGNORE_ITEM item_name IGNORE_ITEM CCSDSVER IGNORE_ITEM CCSDSTYPE IGNORE_ITEM CCSDSSHF IGNORE_ITEM CCSDSAPID IGNORE_ITEM CCSDSSEQFLAGS IGNORE_ITEM CCSDSSEQCNT IGNORE_ITEM CCSDSLENGTH # Explicitly define command and telemetry files # If not given then all the files in cmd/tlm will be processed in # alphabetical order # COMMANDS inst_cmds.txt # TELEMETRY inst_tlm.txt # TELEMETRY inst_tlm_override.txt
```

Obviously, in this case I am simply using a random file target.txt, but the fact that it works from the Command Sender but can't be re-executed or scripted is a problem. Of course, this is very possibly user error, but if not please correct to bug.

ryanmelt, 2022-08-22:
Sending commands with binary data requires using the alternative commanding syntax.  
Binary data can contain quotes and null characters and commas, that don't work with the single string syntax.

https://openc3.com/docs/v5/scripting#cmd

The alternative syntax looks like this:
cmd("TARGET_NAME", "COMMAND_NAME", "PARAM1" => value1, "PARAM2" => value2)
The values can be whatever and binary data should work just fine.
Let me know if that gets you going.

tj-scherer, 2022-08-23:
That did resolve the issue, perhaps the command history should use this format instead? It seems that this format will ensure the correct behavior whereas the cmd("TARGET with PARAM #") format does not.

ryanmelt, 2022-08-24:
The regular syntax is easier for less technical users.  We will consider a toggle though.

closed 2022-08-24 by ryanmelt