OpenC3/cosmos #93: How do I send a command to an instrument from arbitrary Ruby code?

bradley-harden-apl, 2022-08-24:
@ryanmelt and @jmthomas, are you still following issues and discussion on the COSMOS repository at all? Are there any plans to migrate issues or redirect users?

In any event, I was hoping to get a response to [my colleague's question](https://github.com/BallAerospace/COSMOS/discussions/1774). Is there a way to send commands from a PROCESSOR?

ryanmelt, 2022-08-24:
We're hoping to be able to take responsibility for the COSMOS repositories again within a month or so.  Until then, we're only responding here.

Jason did already respond to this particular question here: https://github.com/OpenC3/openc3/issues/88

Please consider purchasing a support contract and supporting our company.

closed 2022-08-24 by ryanmelt

bradley-harden-apl, 2022-08-24:
Whoops, sorry. I must not have refreshed the page or something. Sorry for the spam