OpenC3/cosmos #105: Best way to run openc3cli bridge command

jurgensymynck, 2022-08-29:
I successfully ran the Ethernet to Serial communication setup to communicate with an Arduino over serial/USB cable, which was explained in  [COSMOS5 ticket #1387](https://github.com/BallAerospace/COSMOS/issues/1387), 

The last command is running  `ruby.exe .\openc3\bin\openc3cli bridge` in the openc3-5 home folder, each time the docker containers are started.
Simple question: where/in what file should I put this command so that it is ran each time the docker containers are up and running?

greetings, Jurgen

ryanmelt, 2022-08-29:
Hi Jurgen,
What is the operating system of your host computer?

jurgensymynck, 2022-08-30:
That's Windows 10 (Enterprise), running Docker Desktop 4.11 and Openc3 5.0.8

closed 2022-08-31 by ryanmelt