OpenC3/cosmos #106: Getting 502 bad gateway error

jgrun, 2022-08-29:
When I run the demo included in OpenC3 5.0.8 I can only get it to reliably launch in Chrome and not Edge. This isn't really a problem, the real problem is that after about 6-10 seconds the telemetry stream in the demo stops and I get repeated bad gateway 502 messages. Any ideas as to why this may be happening?

Due to IT restrictions I am running Docker through Hyper-V, but otherwise I have followed all of the installation instructions provided.

ryanmelt, 2022-08-29:
502 Bad Gateway means that our Traefik reverse proxy can't talk to the OpenC3 backend (in this case probably the openc3-cmd-tlm-api container).

Couple first debugging steps:
1. Run "docker ps" and see if any containers are restarting - you can see if they have small uptimes
2. Run "docker logs openc3-openc3-cmd-tlm-api-1" and look for any crashing type backtraces.  

Please let us know what those show.

jgrun, 2022-08-29:
I noticed that these two highlighted containers are restarting regularly.
[image]
[cmd_tlm_logs.txt](https://github.com/OpenC3/openc3/files/9446788/cmd_tlm_logs.txt)
I've attached what the log dumps. It looks to me like it's running fine for a bit and then it does something with Puma and just keeps rebooting

ryanmelt, 2022-08-29:
It looks like several things are cycling.  My guess is that Docker isn't getting enough resources.  The Hyper-V version is very limited by default. 

If you go into the Docker Desktop app -> Click on the bug icon in the corner, there is a section on resources.  Make sure Docker has at least 8GB of RAM, at least a couple CPU cores, and 60+GB or Disk.

jgrun, 2022-08-29:
That was the problem exactly. I upped the resources for Docker and things are good now. Thanks so much for the help Ryan!

closed 2022-08-29 by jgrun