OpenC3/cosmos #108: When is the planned release date/timeline for OpenC3 Version 5.0.9?

GTI-Dakota, 2022-08-30:
Hello,

My team is anxiously awaiting the next OpenC3 version. Do you have a rough estimate for when to expect that?

Thanks,

Dakota

ryanmelt, 2022-08-30:
Hi Dakota,
We are there on the development front.  I just want to get more testing in on the local mode pull request so that that change makes it into the release.   Should be ready by the end of the day tomorrow (Aug 31st).

GTI-Dakota, 2022-08-30:
Awesome, thanks for the update!

closed 2022-08-31 by ryanmelt

ryanmelt, 2022-09-01:
Hey Dakota - 
Just an update.  This is slipping to tomorrow so we can get all the tests green again.
Local mode itself is fully tested though and I'm excited to get that released.  :)

ryanmelt, 2022-09-02:
https://github.com/OpenC3/openc3/releases/tag/v5.0.9

ryanmelt, 2022-09-02:
Enterprise should be released this afternoon