OpenC3/cosmos #113: [Telemetry Viewer] Limits-related widgets cause TypeErrors

jwestra-carrastro, 2022-08-30:
When a screen contains a LIMITSBAR, VALUELIMITSBAR, etc. widget, e.g. "LABELVALUELIMITSBAR INST TEMPS DIODE_1_TEMP", the screen will always display the limits bar in the RED_LOW region, no matter what the telemetry value is (see below).
[image]

Then, when one opens the built-in screen editor and closes it, the following error occurs:
 (error) TypeError: Cannot read properties of undefined (reading '0')

This issue does not appear to occur in the demo, only in custom targets.

jmthomas, 2022-08-30:
Have you ever received any packets for that telemetry? It could be relate to stale values.

jwestra-carrastro, 2022-08-30:
I am receiving packets that this telemetry item is contained in. This item in particular is an unused UINT16 value that is always 0, and I have labeled it as a temperature and added limits values to for demonstration purposes.
Could the issue be that the limits bar doesn't update from a default position of :RED_LOW until the value changes?

ryanmelt, 2022-08-31:
Limits are checked every time a packet is received.   Can you share the relevant configuration file lines for the telemetry packet?

jwestra-carrastro, 2022-08-31:
targets/INST/cmd_tlm/tlm.txt:
TELEMETRY INST TEMPS BIG_ENDIAN "Temperature Packet"
  APPEND_ITEM    DIODE_1_TEMP 32       FLOAT     "Diode 1 Temperature"

targets/INST/screens/temp.txt:
SCREEN AUTO AUTO 0.5
VERTICAL
    VALUELIMITSBAR INST TEMPS DIODE_1_TEMP
END

PR #250 (2022-11-16): Fix edit screen and 0 values
closes #113
closed 2022-11-16 by jmthomas