OpenC3/cosmos #120: Mounting error help

tyhill2, 2022-08-31:
My containers creation gets to the very end, and then I am receiving this error:

```
#32 [docker.io/openc3inc/openc3-init:latest] exporting to image
#32 exporting layers done
#32 writing image sha256:4191b3f3390c01dc7e8941367132d07acf5f76c4057227c7778480ed4483bdea done
#32 naming to docker.io/openc3inc/openc3-ruby:latest done
#32 writing image sha256:12edd8e3173158d268e6314ff3867a32c0c22157ddcb895590211c46dcdcc00e done
#32 naming to docker.io/openc3inc/openc3-init:latest done
#32 DONE 0.4s
Network openc3_default  Creating
Network openc3_default  Created
Container openc3-openc3-minio-1  Creating
Container openc3-openc3-redis-ephemeral-1  Creating
Container openc3-openc3-redis-1  Creating
Container openc3-openc3-minio-1  Created
Container openc3-openc3-redis-1  Created
Container openc3-openc3-redis-ephemeral-1  Created
Container openc3-openc3-cmd-tlm-api-1  Creating
Container openc3-openc3-operator-1  Creating
Container openc3-openc3-script-runner-api-1  Creating
Container openc3-openc3-traefik-1  Creating
Container openc3-openc3-traefik-1  Created
Container openc3-openc3-init-1  Creating
Container openc3-openc3-operator-1  Created
Container openc3-openc3-script-runner-api-1  Created
Container openc3-openc3-cmd-tlm-api-1  Created
Container openc3-openc3-init-1  Created
Container openc3-openc3-redis-ephemeral-1  Starting
Container openc3-openc3-minio-1  Starting
Container openc3-openc3-redis-1  Starting
Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/run/desktop/mnt/host/wsl/docker-desktop-bind-mounts/Ubuntu-22.04/0bb280bb5603acbb2d29121785d93ec8479975ddfe2297bd5d17b64f6003e2f1" to rootfs at "/devel/cacert.pem": mount /run/desktop/mnt/host/wsl/docker-desktop-bind-mounts/Ubuntu-22.04/0bb280bb5603acbb2d29121785d93ec8479975ddfe2297bd5d17b64f6003e2f1:/devel/cacert.pem (via /proc/self/fd/14), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
```

I am currently in the openc3 folder after I have done a `git clone` from master.
Here is what my Dockerfile looks like:
```
FROM docker

RUN apk add --update docker-compose curl openrc docker

WORKDIR /openc3/

COPY ./ ./

COPY cacert.pem /devel/cacert.pem
ENV SSL_CERT_FILE=/devel/cacert.pem
ENV CURL_CA_BUNDLE=/devel/cacert.pem
ENV REQUESTS_CA_BUNDLE=/devel/cacert.pem

EXPOSE 80

ENTRYPOINT ["/openc3/openc3.sh","start"]
```

Any help on this one?

ryanmelt, 2022-08-31:
Looks like you are trying to do Docker in Docker?   That's not something we have tried.

tyhill2, 2022-08-31:
I am open to other suggestions of how to do it better, but the ask of my company is to just get this running on a hosted VM/Container somehow. I asked about AKS on here yesterday since we currently have that in our Azure cloud, and have emailed sales@openc3.com about it.

ryanmelt, 2022-08-31:
Can you just run locally?

```
# Use our released containers - Fast
./scripts/linux/openc3_setup.sh
openc3.sh run 
```

or

```
# Build and run
openc3.sh start 
```

ryanmelt, 2022-08-31:
You should be able to do that easily on any linux VM, with docker and docker-compose installed.
You can also use podman on RHEL8.

tyhill2, 2022-08-31:
So would you recommend I spin up a VM, git clone master, install docker and docker-compose, and then run the steps above?

ryanmelt, 2022-08-31:
Yep, that should work.  Let us know if you run into any snags.

ryanmelt, 2022-08-31:
We'll also have a stable 5.0.9 release later today, but master should work if you are experimenting.

tyhill2, 2022-08-31:
Okay, thank you!

ryanmelt, 2022-08-31:
Oh - One snag you will run to, is our default security settings don't allow connecting externally unless you configure SSL.
The quickest fix is to open to http, but obviously insecure
You can also provide your own SSL certificates, or generate them using letsencrypt (with an associated domain name pointed at the VM)

tyhill2, 2022-08-31:
Okay. We have our own SSL, so I will provide that. Is there documentation on how to get that done?

ryanmelt, 2022-08-31:
The best documentation is currently here:
https://github.com/OpenC3/openc3-project/blob/main/README.md

This is for the project template "way of running" which will be the standard after our next release, but should still work with cloning the master branch.

tyhill2, 2022-08-31:
For this part:
```
Edit compose.yaml
Comment out this openc3-traefik line: - "./openc3-traefik/traefik.yaml:/etc/traefik/traefik.yaml:z"
Uncomment this openc3-traefik line: - "./openc3-traefik/traefik-ssl.yaml:/etc/traefik/traefik.yaml"
Uncomment this openc3-traefik line: - "./openc3-traefik/cert.key:/etc/traefik/cert.key"
Uncomment this openc3-traefik line: - "./openc3-traefik/cert.crt:/etc/traefik/cert.crt"
Edit the openc3-keycloak section and change KC_HOSTNAME to your domain and KC_HOSTNAME_PORT to -1
```

Where is the openc3-keycloak section?

tyhill2, 2022-08-31:
After following all of the directions here: https://github.com/OpenC3/openc3-project/blob/main/README.md

I can successfully get the project to launch on my domain, but it still defaults to using the `TRAEFIK DEFAULT CERT`

ryanmelt, 2022-08-31:
That means you are close. :)
Traefik can be a little finicky.   Make sure your domain is filled in in the traefik configuration file, and that your cert files are in the correct place.

tyhill2, 2022-08-31:
Do the env vars needs to be updated?
```
SSL_CERT_FILE: "/devel/cacert.pem"
CURL_CA_BUNDLE: "/devel/cacert.pem"
REQUESTS_CA_BUNDLE: "/devel/cacert.pem"
NODE_EXTRA_CA_CERTS: "/devel/cacert.pem"
```

ryanmelt, 2022-08-31:
No, the cacerts are just if you need to do things to work around company internal certificates to get out to like the alpine package repo.

You do need to put cert.crt and cert.key in the right place.

tyhill2, 2022-08-31:
Currently they are in ./openc3-traefik/
```
~/openc3/openc3-traefik$ ls
Dockerfile      Dockerfile-dev-base  cert.crt  cert.pem                 traefik-dev-base.yaml  traefik-letsencrypt.yaml  traefik.yaml
Dockerfile-dev  cacert.pem           cert.key  traefik-allow-http.yaml  traefik-dev.yaml       traefik-ssl.yaml
```

tyhill2, 2022-08-31:
What about my `keycloak` question above?

ryanmelt, 2022-08-31:
Keycloak is an enterprise feature, that is a reference that shouldn't be there.

cert files look like in the right place.  
Did you uncomment the corresponding lines in compose.yaml?
Are you sure .crt is the public key. and .key is the private key?
Do the certs reference the exact correct domain name?

tyhill2, 2022-08-31:
I used this to transform the `cert.pem` file to `cert.crt` and I'm not sure if that matters or if you want .pem vs. .crt
```
openssl x509 -outform der -in cert.pem -out cert.crt
```

ryanmelt, 2022-08-31:
compose looks good.  I don't think you transform the pem.  Just rename to .crt

tyhill2, 2022-08-31:
Ah okay, let me try that

tyhill2, 2022-08-31:
That fixed it! Thank you very much.

closed 2022-09-01 by ryanmelt