OpenC3/cosmos #121: screens and directory hierarchy

steven-efsi, 2022-08-31:
Screens are placed in PLUGIN/screens directory, but they can refer to any target's packet fields. This makes it confusing for organizational purposes. In our case, we have a single plugin for all flight software applications, each a separate target. We would prefer to have two groups work on screens and scripts for these application targets as separate plugins, for different phases of the project. We can create "dummy targets" in these plugins with screens that refer to the application target packet field values, and it works. But, you can't go to "Packet Viewer" and open one of those screens from that target name - -you have to use the name of the "dummy target."

Is there a better way to organize this? Is it intended we do what we are doing where a target maybe has no interface or commands/tlm, but just has procs and screens that act on other targets?

Thanks!

ryanmelt, 2022-08-31:
Hey Steven,

The idea was that you could have targets that just had support information and no cmds/tlm of their own.   Lots of times we have a "SYSTEM" target that contains screens and procedures that are across targets.

It sounds like the ask here is for a plugin that could "augment" an existing target.   That's an interesting idea, and it would definitely add some syncing and "which plugin wins" kinds of problems.

The goal of target plugin is to capture everything you need that is unique to that target.   Could you do what you are doing now, and then in later phases merge the two plugins into one with the ideal target name?

steven-efsi, 2022-08-31:
Hi Ryan. I was just at smallsat with DCM who says he knows you. Thank you for confirming this was your idea. We can move forward with that for now, and yes we can support a merge if the augment idea ever takes off.

closed 2022-09-01 by steven-efsi