OpenC3/cosmos #122: routers

steven-efsi, 2022-08-31:
I was under the impression a ROUTER could be created for say, a udp interface, and then "ROUTE" command under that to route an interface to it.

1) The ROUTE command doesn't seem to exist.

2) The target, intending to be routed, can't use cmd_tlm_interface.rb (doesn't exist anymore) so what would it use? I need one set of UDP ports (for now) for tlm and cmd.

This question came up when I got the visiona CFDP engine working in OpenC3. I wound up having to scrap the "background_task/microservice" idea any instead do it all in ruby threads behind the scenes. This all lives in the same interface as all targets on the UDP interface. This is where the COSMOS5 issue on the github page came up about BLOCK data types sending json_class information when issued with the cmd() function in Ruby. I commented that the bug does not exist in openc3. (though, I could crash the logging pane and have it go completely blank with the code as shown depending on the data I passed!)

ryanmelt, 2022-08-31:
Routers work similar to how they used to, except they map to a target now instead of an interface
You can certainly have a UDP router.

steven-efsi, 2022-09-01:
Well I guess I need to figure out how they used to work. :) But the documentation still references ROUTE with an interface, so that's why I was confused.

steven-efsi, 2022-09-07:
So in my case, I have a single (for now) udp interface to the command & data handling for my flight processor. I would like the CFDP engine to route packets sent and received through this interface, but I'd like the engine to be a separate interface. How would I set this up? Right now the engine is all in one interface so it can receive the packets. The cfdp engine uses a read and write protocol to intercept packets for it and inject them into the ruby environment that processes them.

ryanmelt, 2022-09-08:
Hi Steven - Happy too setup a support contract and help you design your architecture.  Please reach out to sales@openc3.com.