OpenC3/cosmos #135: Bug: Bit fields does not work for little endian format

rashuky, 2022-09-12:
**Describe the bug**
When defining the following telemetry structure with bit fields

The plugin is built successfully, but OpenC3 fails to import it.

**To Reproduce**
The easiest way to reproduce the bug is to clone my fork of the openC3 repository, where I have provided a test that reproduces a bug. Refer to diff to see the provided test: https://github.com/OpenC3/openc3/compare/master...rashuky:openc3:bug_little_endian_bit_fields 

1. Clone https://github.com/rashuky/openc3/tree/bug_little_endian_bit_fields 
2. Change folder `cd openc3/openc3`
3. Run test `bundle exec rspec spec/packets/parsers/packet_item_parser_spec.rb`
4. Observe the error
```
OpenC3::ConfigParser::Error:
       APPLICATION-PROCESS-IDENTIFIER: LITTLE_ENDIAN bitfield with bit_offset 0 and bit_size 11 is invalid
```

Another way of reproducing a bug would be to build a plugin and import it to OpenC3

1. Create a new plugin `openc3.sh cli generate plugin bug_little_endian`
2. Edit tlm.txt file. 
    1. Go to `cd openc3-bug-little-endian/targets/BUG_LITTLE_ENDIAN/cmd_tlm/`
    2. Open tlm.txt
    6. Replace the content of file with
```
TELEMETRY obc-performance Obc-performance LITTLE_ENDIAN "(0x0899) SANDBOX PC housekeeping telemetry"
    APPEND_ID_ITEM      packet-version-number                 3 UINT          0 ""
    APPEND_ID_ITEM      packet-type-is-cmd                    1 UINT          0 ""
        STATE False                           0
        STATE True                            1
    APPEND_ID_ITEM      sec-hdr-flag-is-present               1 UINT          1 ""
        STATE False                           0
        STATE True                            1
    APPEND_ID_ITEM      application-process-identifier       11 UINT       0x99 ""
```
3. Build the plugin 
    1. Go back to the plugin main folder 
    2. Execute command `openc3.sh cli rake build VERSION=1.0.0`
7. Start openc3: `openc3.sh start`
8. In the browser open localhost:2900
9. Upload the plugin
    1. Go to the admin tab
    2. Download the built plugin
10. The plugin installation should crash. You should observe the following error
```
Stderr:
Error installing plugin: DEFAULT: openc3-bug-little-endian-1.0.0.20220912073345.gem: Error : APPLICATION-PROCESS-IDENTIFIER: LITTLE_ENDIAN bitfield with bit_offset 0 and bit_size 11 is invalid
```
**Expected behavior**
I would expect that the [Little]([url](url)) Endian format is supported for bit fields. 

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS: [e.g. Windows 10 with WSL2]
- OpenC3 Version [e.g. 5.0.7]
- Browser
  -- [x] Google Chrome

ryanmelt, 2022-09-12:
The logic for little endian bit fields is a bit weird.  I need to add a section to the OpenC3 website, but here is an old blog post I wrote on it:
https://ballaerospace.github.io/cosmos-website/news/2018/12/24/little-endian/

Also CCSDS source packets are big endian by definition - You probably shouldn't be defining them as little endian.

rashuky, 2022-09-15:
Ok, now I understand how it works. Thank you, @ryanmelt! It should be mentioned somewhere in the documentation, though. The blog post that you linked explains it pretty well. Maybe it could be integrated somewhere into openC3 documentation?

jmthomas, 2022-09-15:
closed by https://github.com/OpenC3/openc3.com/pull/12

closed 2022-09-15 by jmthomas