OpenC3/cosmos #138: Ability to turn off/reduce logging of cmds from CMD_TLM_API

GTI-Dakota, 2022-09-13:
Hello,

With the new 5.0.9 update, the CMD_TLM_API logs out every command that is sent from any scripts, microservices, or screens. Is there a way to turn this feature off? Performance takes a hit as more command-based microservices and scripts are run.

Thanks,

Dakota

GTI-Dakota, 2022-09-13:
Performance impact can be observed on any window that provides insight to the logger. Additionally, the log window is flooded with continuous messages which makes it extremely hard to pinpoint and read error messages that are generated by microservices and interfaces (even while using the filter option, as the rolling log erases the error message too quickly)

ryanmelt, 2022-09-13:
If you have a background process sending a specific command all the time (for example polling for telemetry), you can disable log messages for specific commands by adding: "DISABLE_MESSAGES" to the command definition.

We're also aware the the scrolling log message window in CmdTlmServer is not ideal.  Plan is to add a new tool for log message analysis, where you can see more messages, filter, and query back in time.  That probably won't make it in until 5.0.11.  I'll go ahead and write the ticket to track though.

GTI-Dakota, 2022-09-15:
Disabling log messages I think is not the way that we'd like to go about it, as some of our commands have states that can determine if we want the command logged or not, and it looks like the DISABLE_MESSAGES turns off logging for all states of a command. This is problematic for commands with hazardous states, or commands that are used for both requesting and setting values (such as in the case of SCPI)

GTI-Dakota, 2022-09-15:
This would also mean we would need to add DISABLE_MESSAGES onto many, many commands across gems, which can be done, but is cumbersome for a user creating command definitions and updating already existing ones

jmthomas, 2022-09-15:
If we disable command logging completely (currently not a feature) you would lose the log messages about hazardous commands and your state commands. However you feel like it's cumbersome to add DISABLE_MESSAGES to many different individual commands. Do you have a recommended solution? We don't know which commands are important so we log everything.

Note that some customers auto-generate their cmd/tlm definitions from a database rather than edit them by hand. So for those people adding a field like "disable logging" would auto-generate the definition with this message. I've always hand edited the definitions and I realize it's a pain but not sure how else to express this information.

GTI-Dakota, 2022-09-15:
Is there a way to update severity/information of the CMD_TLM_API's logger in general? All the commands being logged are INFO, but if we could bump that to WARNING that may resolve the issue. If it can be adjusted to warning, then Hazardous would still log, while general INFO commands would not. Does that sound feasible or am I interpreting the architecture of the logger configuration incorrectly?

GTI-Dakota, 2022-09-15:
Sounds great, thanks!

PR #149 (2022-09-16): Add severity level filter to log messages
closes #138
closed 2022-09-16 by jmthomas