OpenC3/cosmos #141: Data Extractor Performance

GTI-Dakota, 2022-09-13:
Hello,

The Data Extractor has performance issues observed when extracting multiple packets. It seems to freeze and must be exited.

Thanks,

Dakota

ryanmelt, 2022-09-13:
Have you seen this with 5.0.9?   A fix was at least attempted in that release.

GTI-Dakota, 2022-09-13:
Yep, we are currently on build 5.0.9

jmthomas, 2022-09-14:
Is there a reproducible use case based on the demo we can use to verify / test?

GTI-Dakota, 2022-09-15:
I can attempt to recreate this issue w/ the demo project. We are currently utilizing the Data Extractor on a gem with many packets and data over a span of a few hours, and that is where we are seeing our issues.

ryanmelt, 2022-10-01:
I can't get this to recreate.  Is this something you can show on a screen share?

ryanmelt, 2022-10-14:
Dakota - Have you seen this happen again?

GTI-Dakota, 2022-10-17:
Ryan,

Yes, we are still seeing it. Could we figure out a time that works for us
and maybe do a screen share soon?

Thanks,

Dakota

GTI-Dakota, 2022-10-17:
Ryan,

Does anytime tomorrow work for you to help us with the Data Extractor and
another issue with interfaces/protocol behavior between 5.0.11 servers?

Thanks,

Dakota

ryanmelt, 2022-10-17:
Yes - I'm free anytime besides 10-11am.   
Are you trying to extract all of your telemetry packets at once?

ryanmelt, 2022-10-18:
Anytime except 10-11am.

GTI-Dakota, 2022-10-18:
Ryan,

My apologies, I read that wrong. How about 11?