OpenC3/cosmos #142: Server Response 500 Error when server is under heavy command load

GTI-Dakota, 2022-09-13:
Hello,

When running scripts with a lot of cmd or tlm api calls, the script will halt with a "No Response from Server 500 Error". this has not been observed on previous version that we have run on OpenC3. Unfortunately, when this script halts and it is a subscript, when the "GO" button is hit, it exits and the parent script continues, which can be hazardous for operational/functional capabilities if the previous script has not exited in an expected manner.

Apologies for the low quality, but here is an example of said error:
[image]

Thanks,

Dakota

ryanmelt, 2022-09-13:
Is this something you can recreate with our demo project?

GTI-Dakota, 2022-09-15:
I will try to do that later today. Unfortunately, as we have learned since creating this issue, that this 500 Error occurs when microservices call the API as well, and due to this issue being unhandled, the microservice crashes and reboots. We are running multiple microservices and gems on our build, so we are losing microservices pretty regularly due to this 500 error. This 500 can be replicated across multiple different servers running OpenC3 on our side, so I do not believe this issue is localized to one machine.

ryanmelt, 2022-09-15:
Sure, I just need to understand the nature of the errors.   If you are overwhelming the API there are things we can do to increase the max requests per second.   If it's something else, then we'll have to dig deeper.

GTI-Dakota, 2022-09-15:
Is there a way we can increase the max requests per second and see how that goes? We can rebuild a container if needed

ryanmelt, 2022-09-15:
One easy thing to try would be to set these environment variables on the openc3-cmd-tlm-api container
RAILS_MAX_THREADS=20
RAILS_MIN_THREADS=20

Those default to 5.   The bottleneck may be deeper though so I'd like to know more about how you are using the system.

GTI-Dakota, 2022-09-16:
I will try that!

PR #165 (2022-10-01): Refresh tokens 60 seconds before expire
closes #142
closed 2022-10-01 by ryanmelt