OpenC3/cosmos #143: Is there a way to install/load external binaries/libraries outside of the RubyGems method on the admin panel?

GTI-Dakota, 2022-09-13:
Hello,

On a previous version of COSMOS, an external binary could be installed in the operator container by modifying the Dockerfile. This seems to have changed, as gems are loaded and stored differently with OpenC3. Is there a way to do this outside of the RubyGems method?

Thanks,

Dakota

ryanmelt, 2022-09-13:
What is the end goal?   To give an arbitrary binary to a script?  Is it an executable or data?

You can always put anything into a container by altering the Dockerfile and rebuilding, but that is generally not ideal.   The goal is to be able to handle all use cases with plugins.

For individual files, the best way is to put them into a target, and then get them with get_target_file().   

If you need something major like adding in python - Then yes you would need to build a custom container.

GTI-Dakota, 2022-09-15:
We will need to load in libraries that are not supported by a rubygem repository. They are ruby-based binaries and gems that we would like installed into the container when it is built and that method of install seems to have changed from COSMOS -> OpenC3. An example being beta builds of libraries/gems that were provided outside of a repository, that must be integrated to the runtime so it can easily be called via 'require' across multiple gems.

GTI-Dakota, 2022-09-20:
I am closing this issue. We have had success w/ Dockerfile updates. Apologies for the inconvenience!

closed 2022-09-20 by GTI-Dakota