OpenC3/cosmos #144: Log Messages Tool

ryanmelt, 2022-09-13:
Create a new tool that:
1. Scrolls more log messages than CmdTlmServer
2. Allows for filtering down log messages at a query level
3. Allows retrieving historical log messages

MTI-twalker, 2022-11-16:
Please add a pause button as well since debugging with the current scrolling messages is like catching fish with your hand.

ryanmelt, 2022-11-16:
Fish catching has been fixed in changes for the next release (pause button added to CmdTlmServer).   We'll do the same when we get to the dedicated tool.

jeffstjean, 2023-03-02:
Will this also support changing the log level at runtime (eg. switching from the default INFO to DEBUG)?

dlackner, 2023-04-20:
This will be very helpful!

jmthomas, 2025-09-14:
closed by https://github.com/OpenC3/cosmos-enterprise/pull/295

closed 2025-09-14 by jmthomas