OpenC3/cosmos #145: Error running with latest docker-compose

pankakes14, 2022-09-14:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
When running v5.0.9 on my ubuntu system, I'm getting the following error when trying to run with `./openc3.sh run` 

```
ERROR: The Compose file './compose.yaml' is invalid because:
Unsupported config option for services.openc3-cmd-tlm-api: 'group_add'
Unsupported config option for services.openc3-init: 'group_add'
Unsupported config option for services.openc3-operator: 'group_add'
Unsupported config option for services.openc3-script-runner-api: 'group_add'
```

Some quick googling showed that this `group_add` is no longer supported in v3? I haven't explicitly updated my docker-compose, but maybe ubuntu updated something out from under me.

Versions:
```
Docker version 20.10.12
docker-compose version 1.25.0
```

**To Reproduce**
Steps to reproduce the behavior:

1. Run `./openc3.sh run` with latest release and above version of docker-compose

**Expected behavior**
Should run

**Environment (please complete the following information):**

- OS: Ubuntu 20.04
- OpenC3 Version 5.0.9
- Browser
  -- [] Internet Explorer (IE)
  -- [] Google Chrome
  -- [] Chromium based browsers
  -- [x] Mozilla Firefox
  -- [] Safari

jmthomas, 2022-09-14:
According to [this](https://github.com/docker/compose/issues/4700) group_add is supported with 1.27.0+ docker-compose for all v2, v3 versions. Please upgrade your docker-compose to 1.27 or later and let me know if that fixes it.

ryanmelt, 2022-09-14:
group_add is needed to support the new local mode feature.  If if you remove the local mode environment variables from .env, you can also remove the group_add lines (but no local mode). 

We'll update our requirements to list docker-compose 1.27+ going forward.

closed 2022-09-14 by ryanmelt

pankakes14, 2022-09-15:
Ahh ok thanks that's my bad. I didn't check to see whether that was the actual latest release on the site. For some reason ubuntu's `apt` was telling me 1.25 was the latest version. Think I had apt pulling from the wrong repo. Thanks for the info!

pankakes14, 2022-09-15:
I updated docker and docker compose and it does work now. However, one note, I did have to change all instances of `docker-compose` to `docker compose` in the startup scripts. Seems that now compose is a subcommand of docker in the new versions instead of a separate entity.

dmikushin, 2022-12-13:
`group_add` does not work with any `version: '3.*'` setting, and works only with `version: '2.4'`

Please provide an example v3 config, for which `group_add` has been re-enabled :pray:

ryanmelt, 2022-12-13:
See earlier comments.  You need to have docker-compose version 1.27 or newer

dmikushin, 2022-12-13:
I do, but it still doesn't work:

```
> docker-compose --version
docker-compose version 1.29.2, build unknown
```