OpenC3/cosmos #157: Trouble accessing TEXTFIELD from BUTTON code

jhayes-apl, 2022-09-27:
I have created the following screen.  Note the NAMED_WIDGET TEXTFIELD named CH4VOLT and the BUTTON that reads the TEXTFIELD.
<PRE>
SCREEN AUTO AUTO 1.0

VERTICALBOX
  HORIZONTAL
    VERTICAL
      LABEL "Chan"
      LABEL "4"
    END
    VERTICAL
      LABEL "Volts"
      NAMED_WIDGET CH4VOLT TEXTFIELD 12 "5.0"
    END
  END
END

HORIZONTALBOX
  BUTTON 'Set' "api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE '+screen.get_named_widget('CH4VOLT').text())"
END
</PRE>

When I open this in COSMOS (SHOW SCREEN), the TEXTFIELD is not displayed.  It should show "5.0".  This can be seen in the attached screen shot.  Pressing the BUTTON sends the command (with the default 5.0).
[image]

I have discovered that if I click the screen's edit button (the pencil icon), then hit Cancel, the TEXTFIELD appears.  This can be seen in the second screen shot.
[image]

However, now pressing the BUTTON results in an error:
<PRE>
[
    {
        "type": "error",
        "message": "TypeError: screen.get_named_widget(...) is undefined",
        "time": 1664286550115
    }
]
</PRE>

I am using COSMOS 5.0.4 on Linux; the browser is Firefox.

ryanmelt, 2022-09-27:
We've made several fixes to TlmViewer since 5.0.4.  Can you try our 5.0.10 release?

jmthomas, 2022-10-10:
@jhayes-apl have you had a chance to try this in the latest release?

jhayes-apl, 2022-10-10:
It is on my "to do" list.  I will let you know the results.

jhayes-apl, 2022-10-18:
We have switched to OpenC3 5.0.11.  There were two problems originally (5.0.4):
1. The values in a TEXTFIELD do not appear on the screen
2. If you click edit (the pencil), the value would appear, but button code couldn't access the value.

Number 2 is fixed; number 1 persists.  I created three simple screens to illustrate the problem, power4.txt, power5.txt, and power6.txt.  These are shown in the following screen shot; only power5.txt shows the TEXTFIELD values:
[image]
Here are the files, power4.txt:
<PRE>
SCREEN AUTO AUTO 1.0

MATRIXBYCOLUMNS 2
  LABEL "Chan"
  LABEL "V"
  LABEL "#3"
  NAMED_WIDGET CH3VOLT TEXTFIELD 12 "10.0"
  LABEL "#4"
  NAMED_WIDGET CH4VOLT TEXTFIELD 12 "12.0"
END

HORIZONTALBOX
  BUTTON 'Set 3' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE '+screen.get_named_widget('CH3VOLT').text())"
  BUTTON 'Set 4' "api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE '+screen.get_named_widget('CH4VOLT').text())"
END
</PRE>

And, power5.txt (the one that works):
<PRE>
SCREEN AUTO AUTO 1.0

VERTICAL
  HORIZONTAL
    LABEL "Ch"
    LABEL "V"
  END
  HORIZONTAL
    LABEL "#3"
    NAMED_WIDGET CH3VOLT TEXTFIELD 12 "10.0"
  END
  HORIZONTAL
    LABEL "#4"
    NAMED_WIDGET CH4VOLT TEXTFIELD 12 "12.0"
  END
END

HORIZONTAL
  BUTTON 'Set 3' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE '+screen.get_named_widget('CH3VOLT').text())"
  BUTTON 'Set 4' "api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE '+screen.get_named_widget('CH4VOLT').text())"
END
</PRE>

And, power6.txt:
<PRE>
SCREEN AUTO AUTO 1.0

HORIZONTAL
  VERTICAL
    LABEL "Ch"
    LABEL "#3"
    LABEL "#4"
  END
  VERTICAL
    LABEL "V"
    NAMED_WIDGET CH3VOLT TEXTFIELD 12 "10.0"
    NAMED_WIDGET CH4VOLT TEXTFIELD 12 "12.0"
  END
END

HORIZONTAL
  BUTTON 'Set 3' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE '+screen.get_named_widget('CH3VOLT').text())"
  BUTTON 'Set 4' "api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE '+screen.get_named_widget('CH4VOLT').text())"
END
</PRE>

jmthomas, 2022-11-15:
It looks like we're treating that width field (12) as pixels instead of characters. I'll work a fix but for now you can change it to something like 120 and you'll be able to see it.

closed 2022-11-16 by jmthomas