OpenC3/cosmos #159: Can commands in a BUTTON be spread over multiple lines for readability?

jhayes-apl, 2022-09-27:
I am trying to create a screen that controls a power supply (via the TEMPLATE protocol).  The user will fill in the voltages and currents (via TEXTFIELDs).  There will be a button that sends the commands to set up the power supply.  I can put all the commands on a very long line.  To make it easier to read, is it possible to spread commands over multiple lines in the screen file?  I tried some experiments in this screen:
<PRE>
SCREEN AUTO AUTO 1.0

HORIZONTALBOX
  BUTTON 'Test1' 'api.cmd("POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE 4.0"); api.cmd("POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE 5.0")'
  BUTTON 'Test2' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE 4.0'); api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE 5.0')"
  BUTTON 'Test3' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE 4.0');
api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE 5.0')"
  BUTTON 'Test3a' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE 4.0');" +
"api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE 5.0')"
  BUTTON 'Test3b' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE 4.0');" + \
"api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE 5.0')"
  BUTTON 'Test4' "api.cmd('POWER VOLTAGE_SET with CHANNEL 3, VOLTAGE 4.0');\
api.cmd('POWER VOLTAGE_SET with CHANNEL 4, VOLTAGE 5.0')"
END
</PRE>
Here are the results:
- Test1: works, but one long line
- Test2: also works same using different quotes
- Test3 error:  Syntax error ... literal not terminated before end of script
- Test3a: no errors, but second command is not sent
- Test3b: no errors, but second command is not sent
- Test4: error: Syntax error ... literal not terminated before end of script

Any suggestions?
I am using COSMOS 5.0.4 on Linux.

ryanmelt, 2022-09-30:
The way this works in SCREEN definitions has changed in releases since 5.0.4.   I still think it needs to be a single line though.   @jmthomas Can you take a look?

jhayes-apl, 2022-10-03:
I have no clue how this stuff works, but I will throw out an idea anyway.  Can the screen include/require some javascript code?  So, my_screen.txt might have:
<PRE>
REQUIRE my_helpers.js   # includes my_helper_function1, etc.
...
BUTTON 'Test1' 'my_helper_function1()'
</PRE>
This adds REQUIRE to your parser, but the parser wouldn't have to handle multi-line commands.

jhayes-apl, 2022-11-15:
I have noticed that BUTTONs with more than one command issues the commands in a random order each time the button is clicked.  Maybe this is intentional.  If not, I could raise it as a separate issue.

PR #284 (2022-11-25): Tlm viewer env to scripts
closes #159
closes #84
closed 2022-11-25 by jmthomas