OpenC3/cosmos #164: map_target_to_interface removes all protocols from previously mapped interface

GTI-Dakota, 2022-09-29:
Hello,

when utilizing map_target_to_interface(), interface 1 loses all protocol files when a target (with a custom protocol defined) moves to interface 2. This creates an endless error loop of interface 1 restarting due to "unable to load" previously available protocol file. Can a map_target_to_interface() copy binded protocols rather than move them? Or is there a way to define and utilize a custom protocol outside of a target folder so it can be shared across multiple gems and protocols when a map_target_to_interface call occurs?

Thanks,

Dakota

ryanmelt, 2022-09-30:
If you put any custom interface and protocols into a plugin toplevel lib folder (not a target lib folder), I think that will get you working.  I would need a little more info to be sure.

ryanmelt, 2022-09-30:
Also - I don't think this is a bug.  Removing a target will remove any target specific libaries.  If an interface need to support removing a target, then it can not rely on target specific code.

ryanmelt, 2022-10-02:
Closing - let me know if you need help configuring your plugins to support this.

closed 2022-10-02 by ryanmelt