OpenC3/cosmos #172: Can't receive telemetry over UDP

apogeecmb, 2022-10-02:
I've used COSMOS v4 in the past, but I'm new to OpenC3 v5.  I've been testing it out for a project we're working on.  I'm trying to set up a target with a UDP interface to send commands and receive telemetry.  I've been able to successfully send commands out of OpenC3 and receive them in a test python script, but I'm having issues getting telemetry into OpenC3.  Here are the contents of my plugin.txt file.

# Set VARIABLEs here to allow variation in your plugin
# See https://openc3.com/docs/v5/plugins for more information
VARIABLE testproject_target_name TESTPROJECT

# Modify this according to your actual target connection
# See https://openc3.com/docs/v5/interfaces for more information
TARGET TESTPROJECT <%= testproject_target_name %>
INTERFACE <%= testproject_target_name %>_INT udp_interface.rb 172.17.0.1 30000 30001 30002 nil 128 10.0 nil
  MAP_TARGET <%= testproject_target_name %>

I'm running OpenC3 from Linux.  I did have to change the ip address in the interface definition from "host.docker.internal" to 172.17.0.1 to get commands working.  

Here is my short and simple test telemetry packet definition in tlm.txt in my target directory.

TELEMETRY TESTPROJECT STATUS BIG_ENDIAN "Telemetry description"
  # Keyword      Name  BitSize Type   ID Description
  APPEND_ID_ITEM ID    8      UINT    1  "Identifier"
  APPEND_ITEM    BOOL  8       UINT      "Boolean"

I'm sending a dummy telemetry packet (b'\x01\x02') from python to 172.17.0.1:30001, but I'm seeing no indication in OpenC3 that anything is being received, successfully or otherwise.  Am I missing something really simple and obvious?  Thanks for any help you can provide.

ryanmelt, 2022-10-03:
You probably need to add the UDP port to your compose.yaml for the openc3-operator process.

apogeecmb, 2022-10-03:
Ryan, thanks so much!  That was it.  Makes sense now.  I didn't have to add the port over TCP because the container was reaching out to the host to establish the connection as opposed to the host having to send into the container for UDP.  Thanks again.