OpenC3/cosmos #174: Local Mode - Changed script file not used by plugin

vidfare, 2022-10-03:
I'm running the latest OpenC3 (5.0.10).
I made a change in an interface script file in the Script Runner. When I save it it is seen under plugins/DEFAULT/targets_modified/.
I can run the script in the Script Runner. But in the running plugin it looks like the original file is used, not the changed one.
Can the plugin not use the changed script directly? Do i have to re-create the plugin with the changed script to test it in the plugin?

ryanmelt, 2022-10-03:
Interfaces and other microservices use the plugin code files as they are in the original plugin gem file.   (Unless you write code to pull the latest version from minio).

targets_modified is for changing scripts/screens/tables without having to rebuild a plugin.

So, yes, you do have to rebuild a plugin and update to affect microservices.

closed 2022-10-03 by ryanmelt

vidfare, 2022-10-03:
It's unfortunate that you have to rebuild the plugin to test the changed script in a microservice. Something I miss from COSMOS4 is that it was much quicker to test a change. You could either reload the configuration in CmdTlm-server, or restart the CmdTlm-server to test a change. In OpenC3 it takes time to update the plugin and wait for all microservices to start.