OpenC3/cosmos #186: Spawn script as background script

jmthomas, 2022-10-05:
Script Runner API method to spawn a script in the background

MTI-twalker, 2022-11-16:
How is this different than a microservice?

Would it be possible to instead have the admin->microservice tab have start/stop buttons that can refresh running instances of microservices? 
That way you can upload your microservice plugin and control when its running.

ryanmelt, 2022-11-16:
Microservices are long lived and kept alive by our framework, where scripts are short lived.

Being able to cycle microservices from the Admin tab would be a great feature though.

jmthomas, 2022-12-12:
You can run a script in the background using the `script_run` API:

        script_run("INST/procedures/collect.rb")

If the script requires user input it just waits until you connect to it.

closed 2022-12-12 by jmthomas

mkalin2, 2025-01-21:
Is there a function that can stop a script that is running in the background?

I tried using `script_delete` but that didn't seem to work.

jmthomas, 2025-01-21:
You're looking for `running_script_stop(id)`

Once we merge https://github.com/OpenC3/cosmos/pull/1803 this will all be documented

mkalin2, 2025-01-21:
Nice! That works thanks.