OpenC3/cosmos #198: [OpenC3 5.0.11+] How many interfaces and targets can I have running at once, and by how much does splitting them into different plugins improve performance?

jwestra-carrastro, 2022-10-06:
Our project has a number of different test configurations (currently 14) using the same devices for each. To support this, I have a TCP/IP interface for each device (currently there are 4 devices but we will have more) for each test configuration. I have one plugin folder containing a single target folder to control one of the devices, and another containing 3 target folders to control the other 3.

I used an ERB loop to generate 14 different targets from the one target folder, one for each test configuration. I was able to upload this to OpenC3 and control multiple configurations without any issues. The second plugin, on the other hand, crashes my browser while trying to set up all of its 42 targets and 42 interfaces. Before it crashed I saw that many of the microservices were crashing and restarting. Sometimes the cmd-tlm-api container also crashes.

I have several questions concerning scalability:
1. How many targets and interfaces per plugin is recommended?
2. How much RAM is needed per target and interface?
3. How many total TCP/IP interfaces can be connected at once?
4. How many targets and interfaces can OpenC3 have installed at once when they are disconnected? And is storage space for gems and logs the only limiting factor?

jmthomas, 2022-10-10:
How many targets and interfaces per plugin is recommended?

- It's really up to your own architecture. Typically it's a one-to-one mapping but if you have a single physically connection serving multiple logical targets then that's totally fine.

How much RAM is needed per target and interface?

- It's really dependent on the complexity of the cmd/tlm definitions. Our demo 

How many total TCP/IP interfaces can be connected at once?

- There is no limit in OpenC3 itself 

How many targets and interfaces can OpenC3 have installed at once when they are disconnected? And is storage space for gems and logs the only limiting factor?

- Again there is no limit in OpenC3 itself but the limiting factor generally comes down to available RAM. Note that the amount of RAM in your physical system is obviously a limit but the amount assigned to Docker is also a limiting factor. We've deployed a target with thousands of packets containing thousands of items but that was the only target in the system.

Our [Installation Instructions](https://openc3.com/docs/v5/installation) recommend a _minimum_ of 8GB assigned to Docker. My Macbook Air with 16GB easily runs our Demo with only 8GB assigned but in the past I used Windows which required more RAM to perform acceptably. Also note the comment about RAM usage on Windows with WSL2.

If you let your system run for about 15min the RAM usage should stabilize and you can get a good feel for how much memory your are using by doing `docker stats`. I'm using 2.5GiB in the operator container running the Demo which will be the container utilizing the majority of the memory.

Please bear with us as we work to optimize this new architecture going forward. We've spent the last 2 years getting everything built and working correctly and are focusing on optimizing OpenC3 to improve performance with every release.

jwestra-carrastro, 2022-10-19:
We changed our interfaces adding DONT_CONNECT, but with just 28 disconnected interfaces installed, the openc3-operator RAM usage is 9 GB out of 16 GB available to Docker. It looks like the logger microservices are running even when interfaces are disconnected. Is there a way to make these microservices only run when their interface is connected?

jmthomas, 2022-10-19:
What version of OpenC3 are you running? What is your OS? Are you running Docker Desktop directly?

jmthomas, 2022-11-23:
@jwestra-carrastro our 5.1 release is out which addresses a lot of performance issues. Please try there and let me know if you're still having issues.

closed 2022-11-28 by jmthomas