OpenC3/cosmos #201: Can Telemetry Grapher use PACKET_TIME instead of RECEIVED_TIME?

jhayes-apl, 2022-10-07:
I have a secondary/derived telemetry packet:
<PRE>
TELEMETRY INTERFACE_BOARD FGB_DAC_SAMPLE BIG_ENDIAN "Fluxgate board mag. sample"
  # Keyword      Name           BitSize Type   ID      Description
  APPEND_ID_ITEM ID             16      UINT   0       "ID"
  APPEND_ITEM    TIMESTAMP      64      FLOAT          "Time"
    FORMAT_STRING "%+.3f"
  APPEND_ITEM    X              24      INT            "X"
  APPEND_ITEM    Y              24      INT            "Y"
  APPEND_ITEM    Z              24      INT            "Z"
  ITEM PACKET_TIME 0 0 DERIVED "Derived time"
# TBD: produces warning on startup
    GENERIC_READ_CONVERSION_START
      Time.at(myself.read('TIMESTAMP', :RAW))
    GENERIC_READ_CONVERSION_END
</PRE>

I use inject_tlm to generate the derived packet:
<PRE>
  nfifo.times do |i|
    ... details suppressed
    sample = {X:tlm("#{TARGET} #{pkt} X", type: :RAW),
              Y:tlm("#{TARGET} #{pkt} Y", type: :RAW),
              Z:tlm("#{TARGET} #{pkt} Z", type: :RAW),
              TIMESTAMP:t+0.002*i}
    inject_tlm(TARGET, "FGB_DAC_SAMPLE", sample)
  end
</PRE>

This was inspired by:
[(https://github.com/BallAerospace/COSMOS/issues/1276)]

I want my data to be plotted as one sample every 2ms.  However, Telemetry Grapher is using the RECEIVED_TIME.  There is a comment in the issue above that "If you Edit the Data Object you can change Time Item to PACKET_TIME."  The context was COSMOS 4.  Is there a way to do this on COSMOS 5?

jmthomas, 2022-10-10:
This was possible is COSMOS 4 but currently unsupported in OpenC3. I'll make this a feature.

ryanmelt, 2022-11-17:
Our next release uses packet_time.

closed 2022-11-17 by ryanmelt

RossWilliamson, 2023-03-07:
Also hoping for option to plot using PACKET_TIME not RECEIVED_TIME.  v5.5.1 still seems to use RECEIVED_TIME