OpenC3/cosmos #205: Help With Build Custom Widgets

bradenvirdellgenevatech, 2022-10-10:
I have been trying to build out a custom widget but it seems to be having trouble building out the libraries.

Here is is the process I have done so far. What I have been trying to do is take the "BigWidget" out of the example and put it into one of my working plugins.

1. I create a src folder and placed a <widget>.vue file in the folder. The src folder is located directly under the main plugin folder. (in this case its the BigWidget.vue)
2. I create a package.json with the correct build cmd
3. Added "system("yarn run build")" to the Rakefile
4. I run "yarn && yarn build" (seems to get stuck here, specifically says "Building for production library".)
5. I run "./openc3.sh cli rake build VERSION=0.0.0"

Is this the correct procedure or am I missing something? I
If you need more information let me know.
 Any help is appreciated. Thank you.

jmthomas, 2022-10-10:
`yarn run build` executes the build line in the package.json file which is:

      "scripts": {
        "build": "vue-cli-service build --target lib --dest tools/widgets/HelloworldWidget --formats umd-min src/HelloworldWidget.vue --name HelloworldWidget && vue-cli-service build --target lib --dest tools/widgets/BigWidget --formats umd-min src/BigWidget.vue --name BigWidget"
      },

You probably have to update this line.

bradenvirdellgenevatech, 2022-10-10:
So I have tried updating that line to the specific name of the widget I was building and I updated the .vue file of the widget. 
I also updated the plugin.txt to match the new name

For example I changed 
`"scripts": {
    "build": "vue-cli-service build --target lib --dest tools/widgets/HelloworldWidget --formats umd-min src/HelloworldWidget.vue --name HelloworldWidget && vue-cli-service build --target lib --dest tools/widgets/BigWidget --formats umd-min src/BigWidget.vue --name BigWidget"
  },`

to 

`"scripts": {
    "build": "vue-cli-service build --target lib --dest tools/widgets/TestWidget --formats umd-min src/TestWidget.vue --name TestWidget"
  },`

I can get this updated name to build within the demo project without any problems. When I move over to a plugin that I am building within the environment that I have it gets stuck during the "yarn build" still. This leads me to believe that it is some sort of environment issue. Specifically I don't think I have all the requisite files available for it to build properly. What specifically stands out to me is in the package.json file it has a dependency for '"@openc3/tool-common": "5.0.11-beta0"'. Would this dependency possibly be causing the problem?

If I do have to set it up in a specific way what all do I need to create the environment that would allow for the building of these widgets?

ryanmelt, 2022-10-12:
You can try specifying the released version of tool-common and see if that resolves the issue.  Change "@openc3/tool-common": "5.0.11-beta0" to "@openc3/tool-common": "5.0.11"

bradenvirdellgenevatech, 2022-10-17:
I did try your suggestion and found no luck. 
To get around this I was able to build my widgets within the demo folder and move them to my gem. Still not sure why I wasnt able to get my repo to work properly with yarn but this solution is acceptable.

closed 2022-10-17 by bradenvirdellgenevatech