OpenC3/cosmos #212: cmd() blocks when other thread blocks on get_packets()

martinerikwerner-aac, 2022-10-13:
Copy of behaviour-presumed-issue from https://github.com/BallAerospace/COSMOS/issues/1786 which also exists in OpenC3.

When combining get_packets() and cmd() calls in a multi-threaded script, cmd() unexpectedly(?) seems to block while get_packets() is blocking.

I would expect that cmd() called from a separate thread should proceed without blocking regardless of any calls to get_packets()?

I have adapted the reproducing plugin and reproduced the issue with OpenC3.

**To Reproduce**

1. Install the plugin from https://github.com/martinerikwerner-aac/openc3-plugin-cmd-blocking
2. Start the loopback.rb script via script runner
3. Connect the TEST interface
4. Start the test.rb script via script runner
5. Observe the log output

**Expected behavior**

The log output should show regular and frequent timestamps from both the "producer" and the "consumer", and in particular it should not raise a timeout exception.

**Environment**

- OS: Debian 11
- OpenC3 Version 5.0.11
- Browser
  -- [x] Mozilla Firefox

ryanmelt, 2022-10-14:
There are a couple things going on here.
1. I'm going to update the blocking functionality of get_packets to be implemented client side instead of server side.   That will prevent locking up the API across threads.
2. There are other cases where you might run into a similar scenario.  The underlying JsonDRbObject class in the api is built mutex protected so you can only have one API request at a time.  The workaround for that is to create another JsonDRbObject in your other thread which will be able to make requests in parallel with a new connection to the API.

ryanmelt, 2022-11-16:
will be closed by #188

closed 2022-11-16 by ryanmelt