OpenC3/cosmos #213: script runner unable to find script in top-level procedures/ of plugin

martinerikwerner-aac, 2022-10-14:
**To Reproduce**

1. Create and install a plugin with a top-level procedures/ directory which contains a script
2. Go to the script runner and try to open the script

**Expected behavior**

Based on
* The documentation at https://openc3.com/docs/v5/scripting-best-practices#organizing-your-scripts-into-separate-files (partly outdated since mentioning config/ which is no longer accessible?)
* The plugin template Rakefile which includes a top-level procedures/ if it exists

I'm assuming that adding script to this directory is supposed to work, and the script runner should be able to find it?

**Environment**

- OS: Debian 11
- OpenC3 Version 5.0.11
- Browser
  -- [x] Mozilla Firefox

ryanmelt, 2022-10-14:
I think this is mainly a documentation defect. 
procedures folders are only supported under targets.    We do support a top level lib folder for code that crosses targets (or is independent of targets).

martinerikwerner-aac, 2022-10-14:
Ah, ok, I was assuming based on the following line in the template `*.gemspec` file (not `Rakefile`, as I incorrectly said above):
```ruby
s.files = Dir.glob("{targets,lib,procedures,tools,microservices}/**/*") + %w(Rakefile README.md plugin.txt)
```
That the directory was supposed to be "supported" to some degree at least.

Maybe it should be removed there in addition to a potential documentation change?

ryanmelt, 2022-10-14:
Correct. We will remove procedures from that list.

PR #239 (2022-11-07): Remove procedures from gemspec
closes #213
closed 2022-11-07 by jmthomas