OpenC3/cosmos #214: adding dependencies between plugins

martinerikwerner-aac, 2022-10-14:
**Is your feature request related to a problem? Please describe.**

I would like to be able to specify dependencies between plugins, in order to make sure that for example a plugin that consists of a set of test procedures which uses multiple targets from other plugins

**Describe the solution you'd like**

The solution I first expected to work was to just add
```ruby
s.add_runtime_dependency 'openc3-common-target', '~> 1.0.0'
```
to the `*.gemspec` file, but that did not not work and the plugin installed happily even when the dependency was uninstalled.

jmthomas, 2022-10-19:
We noticed that when we install gems we're not logging errors during the gem install. So while it may appear that your gem installed happily it probably did not. This is planned to be fixed in the next release (5.0.12).

ryanmelt, 2022-11-08:
re #241 - This change will ensure that gem dependencies install successfully or the overall plugin install will fail.

Note that this only installs additional "gems" into the ruby ecosystem, it does not install "plugins" which would bring in Interfaces, targets, etc.   Plugins must still be installed individually, though you can reference the need for specific other plugins in this way.  And if you only need the top-level lib folder from another plugin, that will work.

closed 2022-11-08 by ryanmelt