OpenC3/cosmos #216: Storing and presenting previous data with a custom widget?

bradenvirdellgenevatech, 2022-10-17:
The goal of this widget is to take a Telemetry Item which we are receiving as a string x number of times within a give time period and continuously display all received data in rows. 

Pretty much I am trying to create a "live stream" of the incoming data and keep a buffer of pervious data in the same widget much like the log viewer. 

Currently I am able to get the incoming data but have not figured out how to store old data to keep it presented to the user. I have been digging around the code base and it appears the vuex is used as a sort of storage using states to do so. This may be part of the solution.

So I was wondering if there is any way to have an array or string in the custom widgets code base that would be appended to upon receiving new data from the telemetry point so that we can continue to show previously received telemetry?

Any help would be appreciated, thank you.

jmthomas, 2022-10-24:
In our [VWidget.js](https://github.com/OpenC3/openc3/blob/fee17352e6881e95efdf06818e487417319935fa/openc3-init/plugins/packages/openc3-tool-common/src/components/widgets/VWidget.js) code we access our vuex store to get the current value. You'll notice we have a computed value called `_value` which is what calculates the value. You could create a watcher on this value and then store it into a local array. Something like this:

  data() {
    return {
      history: [],
    }
  },
  watch: {
    _value: function (newVal, oldVal) {
      this.history.push(newVal)
      // Add logic to ensure it doesn't fill up forever
    },
  },

bradenvirdellgenevatech, 2022-11-16:
I ended up doing something similar to this, works perfectly.

closed 2022-11-16 by bradenvirdellgenevatech