OpenC3/cosmos #220: Reading raw telemetry via script runner returns converted values (5.0.11)

jhayes-apl, 2022-10-20:
We recently switched to OpenC3 5.0.11.  I have scripts that read telemetry and save the raw values in files for analysis in Excel.  These no longer work correctly; the data returned is converted to engineering units.  Here is the call I use, with some details suppressed:
<PRE>
... = tlm(..., type: :RAW)
</PRE>
This returns converted values.  As an experiment, I changed it to:
<PRE>
... = tlm_raw(...)
</PRE>
This also returns converted values.

We still have COSMOS 5.0.4 on another laptop.  I re-ran a similar script on this laptop and the data returned was raw as expected.  So, I think this problem was introduced somewhere between 5.0.4. and 11.

ryanmelt, 2022-10-21:
Quick test against our demo shows this working just fine:
```
puts tlm("INST HEALTH_STATUS TEMP1")
puts tlm("INST HEALTH_STATUS TEMP1", type: :RAW)
puts tlm_raw("INST HEALTH_STATUS TEMP1")
```

Are you sure RAW and CONVERTED are different for the item you're looking at?

If you right click the value in PacketViewer and show Details, what do you see?

jhayes-apl, 2022-10-25:
Here is the telemetry definition:
<PRE>
  APPEND_ARRAY_ITEM    MAGFGB_X       24      INT    3072    "128 DAC samples"
    POLY_READ_CONVERSION 0 0.007751937984496
    UNITS nT nT
    FORMAT_STRING "%+.3f"
</PRE>
This is what results from "View Raw Items" and right-click Show Details:
[image]
Observations:
- The telemetry item is an array of 128 24-bit values.
- The "raw" values are floating-point numbers.
- If I comment out the three conversion lines, the correct raw values appear (not shown).

PR #247 (2022-11-16): Fix packet decom of array data
closes #220
closed 2022-11-16 by jmthomas