OpenC3/cosmos #221: Plugin IP address, port not visible to local machine

berickso, 2022-10-20:
I am doing some preliminary testing with the OpenC3 software and would like to create a target to send sample data to the software from MATLAB via TCP/IP client.  

I've created the plugin without issue per the documentation provided.  It appears to have been assigned an IP address and port of 192.168.55.2:8082 per the plugin specification and closely resembling the docker desktop default subnet of 192.168.65.0/24, however, the IP and port created cannot be pinged or located via the windows command window nor can it be accessed from a MATLAB TCP client.

Should this target IP, port be immediately visible to the local machine at the time of creation?  I suspect it may not be visible outside of docker due to network configuration which isolates the docker network from the local machine.  Running the docker command "docker network inspect host" it also appears the host network does not have an assigned subnet or gateway.  This is a default installation of docker windows with no further configuration.

Any help is appreciated. 

Thanks,
Brett

ryanmelt, 2022-10-20:
Hey Brett.  You need to open that port to the openc3-operator service in compose.yaml.

berickso, 2022-10-20:
Thanks for the quick response!  Looks like it works, I appended the following to the bottom of the of the openc3-operator portion of the script:

ports:
     - "8082:8082"
     - "8083:8083"

Ports 127.0.0.1:8082 and :8083 are visible on the local network

Is this correct?

Thanks for your help.

jmthomas, 2022-10-20:
Looks good to me. Did it work?

berickso, 2022-10-20:
The target does show it is connected in OpenC3 when running the MATLAB TCP server, so I'd say yes.  I don't yet have a model to send packeted data from MATLAB to test with; more work to get to that point.

Nonetheless, this solved an issue I've been struggling with for a couple days so, big thanks! Network configuration and software is not my background to say the least.

closed 2022-10-24 by jmthomas