OpenC3/cosmos #224: Expose telemetry on an OpenTelemetry endpoint

erdnaxe, 2022-10-28:
**Is your feature request related to a problem? Please describe.**
Currently I use a custom Python script that subscribe to telemetry packets over WebSocket `/openc3-api/cable` and then export the values by serving a webserver with `/metrics` OpenTelemetry endpoint. This enables me to log some useful states to my Prometheus instance, and then do alerting and visualisation with Grafana.
This requires manual work to extract all telemetry packets name (such as `TLM__ADCS__IMU_TLM_PKT__A_X__CONVERTED`) and needs to be updated each time telemetry packet structures change.

As OpenC3 already stores in memory the last value of received telemetry packets, a `/metrics` endpoint could be added exposing these. Basic auth with user password could be used to protect these data (monitoring software such as Prometheus supports that).

**Describe the solution you'd like**
An authentificated `/metrics` endpoint containing last telemetry.

If current telemetry field is `null`, then do not show the value on `/metrics`.
If the telemetry field type is a float, an int, or an enum, then show the value on `/metrics`.
If the telemetry field type is an array, then it could be split using labels (e.g. `my_array_metric{index=0}`).

`TLM__ADCS__IMU_TLM_PKT__A_X` could be exported as `openc3__adcs__imu_tlm_packet__a_x` for example (OpenSatKit ADCS).

**Additional context**
OpenC3 5.0.11

jmthomas, 2022-11-01:
We recently added OpenTelemetry to our Ruby code in https://github.com/OpenC3/openc3/pull/194 to assist with performance tracing so this is on our radar.

ryanmelt, 2023-02-13:
Prometheus support is now an enterprise feature for COSMOS system metrics.
Also - We can't export every telemetry point on a /metrics endpoint.  That would not scale to a large system.