OpenC3/cosmos #230: View "Command Sender" logs in a telemetry screen?

KodiakLiDAR, 2022-11-02:
Hello OpenC3 Team,

Is there a way to view the "Command Sender" logs or "Command and Telemetry Server" logs in a separate window or telemetry screen?

Bonus points if there is a way to increase the list length so that I can scroll all the way back to the first log message (Within the Command Sender or CTS tools, there is a limit to the log list length, preventing older log messages from being seen).

Thank you very much for any and all guidance.

Kindly,
Kodiak

jmthomas, 2022-11-02:
We are tracking this in #144

KodiakLiDAR, 2022-11-08:
thank you @jmthomas. I'll watch for any advancements on this feature request while I continue to work on a solution. I'll you keep you posted with my progress.

closed 2026-10-13 by EmilyRagan