OpenC3/cosmos #231: DISABLE_MESSAGES for a command's STATE

GTI-Dakota, 2022-11-03:
Hello,

Command states include HAZARDOUS flags, could there be an option to DISABLE_MESSAGES as well? For some HTTP commands, we have GET and POST bundled into the same command for brevity, and it would be nice to suppress all the GET calls in the "Log Messages" window but not POSTS, for example.

Thanks,

Dakota

jmthomas, 2022-11-05:
We have [DISABLE_MESSAGES](https://openc3.com/docs/v5/command#disable_messages) for an entire command. Are you asking for that on an individual state of a command?

GTI-Dakota, 2022-11-07:
Yep! The DISABLE_MESSAGES tag has been very helpful so far, but like HAZARDOUS, it would be nice to attribute it to a state

closed 2022-12-08 by jmthomas