OpenC3/cosmos #234: Updating/adding Commands at-runtime

GTI-Dakota, 2022-11-04:
Hello,

Is there any easy way to modify a command (STATES, specifically), or create one after the gem has been installed? My idea is to generate (or update) a command with states based on items external to the gem. The going theory is that a protocol I've created can handle this in its initialize if the command has not been generated yet. I've checked out the source and it looks like you create commands using the System.instance, is this the only way to do what I am thinking?

The flow would be as follows:
 - Install Gem
 - Gem interface object is created, which initializes the protocol
 - Protocol initializer checks for COMMAND_A, if not existent then COMMAND_A is created. If existent, then update COMMAND_A
 - User can now view and use COMMAND_A

Thanks,

Dakota

jmthomas, 2022-11-05:
I don't think this is currently possible due to the various other microservices that need to know all the packet definitions when they start. @ryanmelt?

ryanmelt, 2022-11-05:
You could do this but not in a Protocol.  It would need to be in code constructed with the packet in every microservice that used it. The constructor for a Conversion or Processor would be a good place.

GTI-Dakota, 2022-11-07:
I see. Would this display the updates in the Command Sender GUI?

ryanmelt, 2022-11-08:
It would, if you also updated the the information we store in Redis models

closed 2022-12-12 by jmthomas