OpenC3/cosmos #235: Unknown Packet With Differing Number of ID_ITEMs in Definitions

jeffstjean, 2022-11-05:
**Describe the bug**
If a telemetry packet is defined with _x_ number of `APPEND_ID_ITEM`s and a second telemetry packet is defined lower in the tlm.txt file with _x_-1 `APPEND_ID_ITEM`s, a packet matching the second packet will not be caught and a `<interface> UNKNOWN packet length: <length> starting with: <data>` will be generated.

**To Reproduce**
Steps to reproduce the behavior:

1. Create a new target with a definition similar to below
2. Send a packet to OpenC3 with a payload of `01 02 03`
3. Observe the `<interface> UNKNOWN packet length: 3 starting with: 010203` warning

```
TELEMETRY TEST FOUR_ITEMS LITTLE_ENDIAN "This overrides anything with three ID_ITEMs"
  APPEND_ID_ITEM FIRST  8 UINT 5
  APPEND_ID_ITEM SECOND 8 UINT 6
  APPEND_ID_ITEM THIRD  8 UINT 7
  APPEND_ID_ITEM FOURTH 8 UINT 8

TELEMETRY TEST THREE_ITEMS LITTLE_ENDIAN "This will never get used"
  APPEND_ID_ITEM FIRST  8 UINT 1
  APPEND_ID_ITEM SECOND 8 UINT 2
  APPEND_ID_ITEM THIRD 8 UINT 3
```

Note: if the three item packet is placed first, it will work correctly but the four item packet will not be caught correctly. It seems as though all packets in a target must have the same number of `ID_ITEM`s otherwise there will be issues.

**Expected behavior**
The number of `ID_ITEM`s should not affect other packet definitions. For each packet, it should just check if all `ID_ITEM`s match and either stop with a success or continue until no packet is found. This seems like it would be occurring in the openc3/packets/telemetry.rb file but there doesn't seem to be anything weird going on there (pasted below).

```
target = System.targets[target_name]
if target and target.tlm_unique_id_mode
  # Iterate through the packets and see if any represent the buffer
  target_packets.each do |packet_name, packet|
    return packet if packet.identify?(packet_data)
  end
else
  # Do a hash lookup to quickly identify the packet
  if target_packets.length > 0
    packet = target_packets.first[1]
    key = packet.read_id_values(packet_data)
    hash = @config.tlm_id_value_hash[target_name]
    identified_packet = hash[key]
    identified_packet = hash['CATCHALL'.freeze] unless identified_packet
    return identified_packet if identified_packet
  end
end
```

**Environment (please complete the following information):**

- OS: Ubuntu 20.04.1
- OpenC3 Version: 5.0.9
- Browser
  -- [] Internet Explorer (IE)
  -- [] Google Chrome
  -- [] Chromium based browsers
  -- [x] Mozilla Firefox
  -- [] Safari

ryanmelt, 2022-11-05:
Add TLM_UNIQUE_ID_MODE to your target.txt file.
https://openc3.com/docs/v5/target#tlm_unique_id_mode

closed 2022-11-05 by ryanmelt

jeffstjean, 2022-11-06:
Thanks! Must've missed that. Any idea how much this affects performance? We're trying to support all the services that the Cubesat Space Protocol (CSP) offers but the CMP port has messages with "sub-commands" that are causing this issue. Any idea at what rates we'd expect to start seeing performance issues?