OpenC3/cosmos #245: openc3-init container doesn't start after clean install, minio "Unable to add new policy. Access Denied"

jurgensymynck, 2022-11-12:
**Describe the bug**
After a clean install following the instruction on the documentation website, the Docker container openc3-init fails to start properly and i cannot go to the openc3  screen at http://localhost:2900/

```
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied.</Message>
<Key>base/index.html</Key>
<BucketName>tools</BucketName>
<Resource>/tools/base/index.html</Resource>
<RequestId>1726D3D998B30FA0</RequestId>
<HostId>528f20f3-2e30-4ee0-8cda-43865ee9a3ff</HostId>
</Error>
```

**To Reproduce**
- in the windows C:\ root directory, a command screen: git clone https://github.com/openc3/openc3-project.git openc3-TiSCo
- start docker desktop
- change to c:\openc3-TiSCo, run openc3.bat start
- all containers are built and run successfully, except openc3-init with errors:

***log messages from openc3-init container***
```
2022-11-12 12:37:41 2022-11-12T11:37:41Z waiting for Minio http://openc3-minio:9000 RC: 0
2022-11-12 12:37:42 2022-11-12T11:37:42Z waiting for Redis. RC: 0
2022-11-12 12:37:43 Added `openc3minio` successfully.
2022-11-12 12:38:44 2022-11-12T11:38:44Z waiting for Minio http://openc3-minio:9000 RC: 0
2022-11-12 12:38:45 2022-11-12T11:38:45Z waiting for Redis. RC: 0
2022-11-12 12:38:46 Added `openc3minio` successfully.
2022-11-12 12:37:43 mc: <ERROR> Unable to add new policy. Access Denied.
2022-11-12 12:38:46 mc: <ERROR> Unable to add new policy. Access Denied.
```

***log message from openc3-minio container***
```
2022-11-12 12:37:43 
2022-11-12 12:37:43 API: SYSTEM()
2022-11-12 12:37:43 Time: 11:37:43 UTC 11/12/2022
2022-11-12 12:37:43 DeploymentID: 528f20f3-2e30-4ee0-8cda-43865ee9a3ff
2022-11-12 12:37:43 Error: Access Denied. (*errors.errorString)
2022-11-12 12:37:43        requestHeaders="{\"method\":\"PUT\",\"reqURI\":\"/minio/admin/v3/add-canned-policy?name=script\",\"header\":{\"Accept-Encoding\":[\"zstd,gzip\"],\"Content-Length\":[\"750\"],\"Host\":[\"openc3-minio:9000\"],\"User-Agent\":[\"MinIO (linux; amd64) madmin-go/0.0.1 mc/RELEASE.2022-10-01T07-56-14Z\"],\"X-Amz-Content-Sha256\":[\"e2ac085bbdeda2382635a2fb60ff71b966a56dc563ef053f5d717cd589c5364b\"]}}"
2022-11-12 12:37:43        6: internal/logger/logger.go:259:logger.LogIf()
2022-11-12 12:37:43        5: cmd/auth-handler.go:155:cmd.validateAdminSignature()
2022-11-12 12:37:43        4: cmd/auth-handler.go:166:cmd.checkAdminRequestAuth()
2022-11-12 12:37:43        3: cmd/admin-handler-utils.go:47:cmd.validateAdminReq()
2022-11-12 12:37:43        2: cmd/admin-handlers-users.go:1440:cmd.adminAPIHandlers.AddCannedPolicy()
2022-11-12 12:37:43        1: net/http/server.go:2084:http.HandlerFunc.ServeHTTP()
```

**Environment (please complete the following information):**

- windows 10
- OpenC3 Version 5.0.11
- Browser: Google Chrome
- firewall: windows defender, my wifi network is set to "private profile"
- docker: v4.14.0

jurgensymynck, 2022-11-13:
Hello,

I just finished a brand new Docker Desktop install on Linux Mint 21 (Vanessa, = Ubuntu 22 jammy jellyfish), an I get the exact same problems here...

Am I missing an installation step? I follow the installation guidelines provided here: https://openc3.com/docs/v5/installation 
... and left all provides files intact (like .env)

greetings, Jurgen

**Environment (please complete the following information):**

Linux Mint 21/Ubuntu 22
OpenC3 Version 5.0.11
Browser: Google Chrome
docker: v4.14.0

ryanmelt, 2022-11-13:
You don't need to build the containers for our releases.   Let's eliminate that as a cause.
In .env:
OPENC3_TAG=5.0.11

Then:
./openc3.sh run
or 
openc3.bat run

Note: "start" will build the containers, while "run" uses already built containers

jurgensymynck, 2022-11-13:
Thanks for the reply!

On my Windows machine, changing .env and using the 'run' parameters I get same problem.

Now, I can't access the error message at ```http://localhost:2900``` anymore, but see below an abridged version of the output of ```openc3.bat run```. Note the complaints about unavailable port(s). This error was likely also present with my earlier attempts but I didn't catch it then. 
I got similar error about port 2900, after trying to tell Windows Defender firewall to allow incoming connections to port 2943. More fiddling with Firewall rules proved unsuccessful (for now at least).

The same message appears as a red banner in Docker Desktop when i manually try to start the ```openc3-traefik-1``` container (see screenshot)

Thanks in advance!
Jurgen
```
Microsoft Windows [Version 10.0.19043.1889]
(c) Microsoft Corporation. Alle rechten voorbehouden.

C:\OpenC3-TiSCo>cd ..
C:\>git clone https://github.com/openc3/openc3-project.git OpenC3-TiSCo
Cloning into 'OpenC3-TiSCo'...
remote: Enumerating objects: 83, done.
remote: Counting objects: 100% (83/83), done.
remote: Compressing objects: 100% (56/56), done.
Receiving objects:  83% (68/83)used 59 (delta 20), pack-reused 0 eceiving objects:  68% (57/83)
Receiving objects: 100% (83/83), 152.35 KiB | 2.34 MiB/s, done.
Resolving deltas: 100% (37/37), done.

C:\>cd OpenC3-TiSCo

C:\OpenC3-TiSCo>openc3.bat run
...
[+] Running 11/12
 - Network openc3-tisco_default                       Created    0.8s
 - Volume "openc3-tisco_openc3-gems-v"                Created    0.0s
 - Volume "openc3-tisco_openc3-minio-v"               Created    0.0s
 - Volume "openc3-tisco_openc3-redis-v"               Created    0.0s
 - Container openc3-tisco-openc3-redis-ephemeral-1    Started    6.6s
 - Container openc3-tisco-openc3-minio-1              Starte...  6.0s
 - Container openc3-tisco-openc3-redis-1              Starte...  6.5s
 - Container openc3-tisco-openc3-script-runner-api-1  Started    6.4s
 - Container openc3-tisco-openc3-cmd-tlm-api-1        Started    6.4s
 - Container openc3-tisco-openc3-operator-1           Sta...     5.4s
 - Container openc3-tisco-openc3-traefik-1            Star...    6.4s
 - Container openc3-tisco-openc3-init-1               Created    0.1s
Error response from daemon: 
Ports are not available: exposing port TCP 127.0.0.1:2943 -> 0.0.0.0:0: listen tcp 127.0.0.1:2943:
bind: Er is geprobeerd toegang te verkrijgen tot een socket op een volgens de toegangsmachtigingen niet toegestane manier.

C:\OpenC3-TiSCo> 
```
[image]

jurgensymynck, 2022-11-16:
Hello,

I went the 'edit .env' route, and my startup problems seem to be resolved. 

Thanks!

closed 2022-11-16 by jurgensymynck